Xilinx/KRS #10: Missing dependency when trying to compile the KRS acceleration examples

kimonk, 2021-09-30:
After installing KRS and following the steps described here: 

https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/0_ros2_publisher.html

I get an error when running _colcon build --merge-install_  
The error message is:

"Starting >>> accelerated_doublevadd_publisher
--- stderr: accelerated_doublevadd_publisher                          
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findvitis_common.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "vitis_common", but CMake did not find one.

  Could not find a package configuration file provided by "vitis_common" with
  any of the following names:

    vitis_commonConfig.cmake
    vitis_common-config.cmake

  Add the installation prefix of "vitis_common" to CMAKE_PREFIX_PATH or set
  "vitis_common_DIR" to a directory containing one of the above files.  If
  "vitis_common" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< accelerated_doublevadd_publisher [0.82s, exited with code 1]"

Apparently some dependency is missing from the repository.

GZanni-mkl, 2021-09-30:
+1 - It happened the same to me

vmayoral, 2021-10-04:
Thanks to both for reporting. It took a while for Xilinx to react and accept pending PRs. https://github.com/Xilinx/KRS/blob/main/krs.repos is now updated.

Re-try with it please.

vmayoral, 2021-10-04:
@kimonk, can you give it a try and report if you still encounter issues?

vmayoral, 2021-10-04:
ping @jasvinderkhurana, please confirm and/or resolve as appropriate.

kimonk, 2021-10-04:
It worked for me. Now I can build the example as described in the guide. I'm closing this issue.

closed 2021-10-04 by kimonk

vmayoral, 2021-10-04:
Thanks @kimonk!