Xilinx/KRS #13: Problem when building SD card targeting kv260 (Example 0 of the doc)

GZanni-mkl, 2021-10-05:
Hi @vmayoral , I successfully go through the steps of installation and then tried to build the example 0 in the doc (https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/0_ros2_publisher.html).
I had an error with mixin option while build the ros package targeting kv260: colcon: error: Mixin 'kv260' is not available for 'build' 

I left below the log of the commands executed in the shell.

[image]

Taking a look at the video left as tutorial in the doc it seems also that my generated firmware folder is correct (or at least as the same structure and folders). Do you have any suggestion?

Thank you in advance

GZanni-mkl, 2021-10-06:
Hi @vmayoral , sorry I posted the wrong screenshoot, as I have done some trail and errors. I have installed the dependencies copying this command from the doc "sudo apt-get -y install curl build-essential libssl-dev git wget \
                          ocl-icd-* opencl-headers python3-vcstool \
                          python3-colcon-common-extensions python3-colcon-mixin \
                          kpartx u-boot-tools"

And it terminates without any error, just a warning when selecting an opencl version.

I have sourced the scripts before the first posting and it gave me the same error. 

I have tried your command "colcon mixins show" but the command failed, as I have in the colcon options "mixin", I do not know whether it was a tapping mistake or it must be called mixins and not mixin.
The command "colcon mixin show" ran, but without any output. In the acceleration list I have enlighted the kv260, as the doc explains.
May I ask if there is another way for debugging? Thank you in advance for your fast answers

GZanni-mkl, 2021-10-09:
Hi @vmayoral ,

I have solved the issue.
The problem was my setup, and it was related (probably) to the fact that I have downloaded in the same directories different versions of the repos from the one needed (this thing was related to this issue:  https://github.com/Xilinx/KRS/issues/10). 
After removing the parent directory of the ws and then re-create it everything went ok. As long as the repo has been updated to solve the issue, the problem doesen't persists anymore, so I close the issue.

Just two advices if I may, the command given to unmount/mount the image:

```bash
pv <your-path-to>/krs_ws/acceleration/firmware/select/sd_card.img | sudo dd of=/dev/rdisk2 bs=4m  # dd the image
```
was not present in my native ubuntu installation (pv) so I reccommend to put it on the dependencies to install (in the step how to install KRS). Second is that the second part of the command should be written as:

```bash
pv <your-path-to>/krs_ws/acceleration/firmware/select/sd_card.img | sudo dd of=/dev/rdisk2 bs=4M  # dd the image
```

with capital M, otherwise the system does not recognize the alignment.

Thank you for your support @vmayoral .

closed 2021-10-09 by GZanni-mkl

vmayoral, 2021-10-10:
Thanks for the feedback @GZanni-mkl, addressed at https://github.com/Xilinx/KRS/pull/15.