Xilinx/KRS #16: KRS alpha release is wrong

vmayoral, 2021-10-12:
@jasvinderkhurana <ins>this has been around for a few weeks now and nobody with privileges has reacted just yet</ins>. KRS was released a month ago and still, other than in the fork that I'm maintaining, the official repo (this one) does not have a proper `alpha` release.

Let's correct the KRS `alpha` release. **Current one is wrong**. Please use content from https://github.com/vmayoral/KRS/releases/tag/alpha. Feel free to retag things at the current HEAD of the repo and re-release it adding https://github.com/Xilinx/KRS/blob/main/krs.repos to the release files for visibility.

jasvinderkhurana, 2021-10-12:
@vmayoral Sorry, for the delay! I just did the alpha release!

jasvinderkhurana, 2021-10-13:
@vmayoral , this is really strange we both are getting different versions. Let's sync up once to fix this.