Xilinx/KRS #24: tracetools_acceleration not found

syed-ahmed, 2021-11-04:
I'm getting this error with Example 0. I'm running on Ubuntu 20.04 and followed the documentation. Is there anything missing in the docs?
```
Starting >>> vitis_common
Starting >>> acceleration_firmware
Starting >>> colcon-acceleration
Starting >>> publisher_xilinx
Starting >>> ros2acceleration
Finished <<< colcon-acceleration [0.36s]                                                                                                                                                            
Finished <<< ament_vitis [0.42s]                                                                                                                                                              
Finished <<< vitis_common [0.42s]
Finished <<< acceleration_firmware [0.42s]
Starting >>> acceleration_firmware_kv260
Starting >>> faster_doublevadd_publisher
Finished <<< acceleration_firmware_kv260 [0.31s]                                                                                                                                    
Starting >>> accelerated_doublevadd_publisher
Starting >>> doublevadd_publisher
Starting >>> offloaded_doublevadd_publisher
Starting >>> simple_adder
Starting >>> triplevadd_publisher
Starting >>> vadd_publisher
Finished <<< ros2acceleration [0.73s]
Finished <<< simple_adder [1.19s]                                                                                                                                                                        
--- stderr: faster_doublevadd_publisher
CMake Error at CMakeLists.txt:20 (find_package):
  By not providing "Findtracetools_acceleration.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "tracetools_acceleration", but CMake did not find one.

  Could not find a package configuration file provided by
  "tracetools_acceleration" with any of the following names:

    tracetools_accelerationConfig.cmake
    tracetools_acceleration-config.cmake

  Add the installation prefix of "tracetools_acceleration" to
  CMAKE_PREFIX_PATH or set "tracetools_acceleration_DIR" to a directory
  containing one of the above files.  If "tracetools_acceleration" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< faster_doublevadd_publisher [1.51s, exited with code 1]
Aborted  <<< triplevadd_publisher [1.38s]                                                                                                                                
Aborted  <<< doublevadd_publisher [1.43s]
Aborted  <<< vadd_publisher [1.44s]
Aborted  <<< accelerated_doublevadd_publisher [1.49s]                                                                                                                           
Aborted  <<< offloaded_doublevadd_publisher [1.49s]
Aborted  <<< publisher_xilinx [3.99s]                                  

Summary: 7 packages finished [4.17s]
  1 package failed: faster_doublevadd_publisher
  6 packages aborted: accelerated_doublevadd_publisher doublevadd_publisher offloaded_doublevadd_publisher publisher_xilinx triplevadd_publisher vadd_publisher
  2 packages had stderr output: doublevadd_publisher faster_doublevadd_publisher

```

jasvinderkhurana, 2021-11-09:
@syed-ahmed, We did not face this issue, but we will recheck this and will get back to you soon.

surender009, 2021-11-10:
Can you please source below paths and try again.

source /tools/Xilinx/Vitis/2020.2/settings64.sh
source /opt/ros/foxy/setup.bash
export PATH="/usr/bin":$PATH 

colcon build --merge-install

vmayoral, 2021-11-11:
@syed-ahmed, the install instructions don't ship with [`tracetools_acceleration`](https://github.com/ros-acceleration/tracetools_acceleration) ([install instructions pointer](https://github.com/Xilinx/KRS/blob/main/sphinx/source/docs/install.md)), so it's normal you're getting that issue if you're trying to build [`acceleration_examples`](https://github.com/ros-acceleration/acceleration_examples/tree/main) (`main` branch)][^1].

You should be targetting [`0.2.0`](https://github.com/ros-acceleration/acceleration_examples/tree/0.2.0) which is what was released in `alpha` and ready to play with.

If you instead prefer to try things in their latest stage ([`main`](https://github.com/ros-acceleration/acceleration_examples/tree/main) branch), then fetch into your workspace (under `src` folder).

[^1]: Note we [updated documentation a while ago](https://github.com/Xilinx/KRS/commit/515c1553dadec4f82e9bf37e3ceb0046dd4ad5b4) to set the `acceleration_examples` to the right branch when cloning. It was wrong before. Maybe this was your issue?

syed-ahmed, 2021-11-11:
aah thanks @vmayoral! That makes sense. I did clone main,. Will try the release branch and I think I have enough to close this issue.

closed 2021-11-11 by syed-ahmed