Xilinx/KRS #28: Update colcon-acceleration in alpha release to include language fixes

vmayoral, 2021-11-23:
Follows from https://github.com/ros-acceleration/community/issues/1#issuecomment-976195585.

Mainly:
- Current KRS alpha ships colcon-acceleration [0.2.0](https://github.com/ros-acceleration/colcon-acceleration/commit/9ff8ca2e2f011759a28e5570fb093d299941f671)
- Changes requested are included in [0.3.0](https://github.com/ros-acceleration/colcon-acceleration/commit/adfc44d0ea37a3df43743e635195548a6b5123b6)

In other words, we need to switch `colcon-acceleration` to tag `0.3.0` in the instructions and krs.repos files. 

@jasvinderkhurana this is an easy release process. Would you like to try it out yourself? Feel free to point me out to the PRs if you need a review.

jasvinderkhurana, 2021-11-23:
@vmayoral 

Sure ! I will work on this and send the PR for review to you.

jasvinderkhurana, 2021-11-23:
@vmayoral I created the new [pull request](https://github.com/Xilinx/KRS/pull/29), but somehow I am unable to add you in reviewer's list. Can you please check if the changes are ok? and we can sync up to resolve the permissions issue.

jasvinderkhurana, 2021-11-23:
@ryanvergel, I am not able add victor as reviewer to Pull Request and neither Victor is able to reopen any KRS issue tickets, can you please check if you can provide the permissions.

jasvinderkhurana, 2021-12-01:
@ryanvergel, any update on this?

closed 2022-09-13 by jasvinderkhurana