Xilinx/KRS #30: AttributeError: 'NoneType' object has no attribute 'split'

tomasthoresen, 2021-11-23:
Getting an error when trying to follow the example.

> colcon acceleration select kv260

ros2@ros2-Latitude-E7450:~/krs_ws$ colcon acceleration select kv260
[0.411s] ERROR:colcon:colcon acceleration: 'NoneType' object has no attribute 'split'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/command.py", line 528, in verb_main
    rc = context.args.main(context=context)
  File "/home/ros2/krs_ws/install/lib/python3.8/site-packages/colcon_acceleration/subverb/select.py", line 57, in main
    firmware_options = get_firmware_options()
  File "/home/ros2/krs_ws/install/lib/python3.8/site-packages/colcon_acceleration/subverb/list.py", line 38, in get_firmware_options
    firmware_options = outs.split("\n")
AttributeError: 'NoneType' object has no attribute 'split'

jasvinderkhurana, 2021-11-23:
@tomasthoresen , can you please clean your workspace and build again

i.e. delete build, install, log, and acceleration folder in krs_ws, then give the build command

```
colcon build --merge-install
source install/setup.bash
colcon acceleration select kv260

jasvinderkhurana, 2021-12-01:
@tomasthoresen , Was your issue resolved shall I close the ticket?

jasvinderkhurana, 2022-09-13:
Closing the ticket!

closed 2022-09-13 by jasvinderkhurana