Xilinx/KRS #34: link Vitis 2020.2.2 outdated

femust, 2022-01-29:
Hey,

there is a minor issue in the instructions
```
- the Vitis `2020.2.2` suite (Vitis, Vivado, Vitis HLS) (`install instructions <https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/vitis/2020-2.html>`_)
```

a link points to the "blank website"

[image]

vmayoral, 2022-01-31:
This sounds about right, https://www.xilinx.com/support/download.html doesn't show Vitis 2020.2.2 anymore. Only Vitis 2020.3 seems available but there's no KRS tested release for this.

@jasvinderkhurana and @giuliocorradi deferring to you on this.

giuliocorradi, 2022-01-31:
Vitis as many other Xilinx tools follow a rolling update - you can find 2020.2 here https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/vitis/archive-vitis.html

femust, 2022-01-31:
Yeap, I figured this out :) 

I guess for future newcomers it would be nice to have this info since at the beginning it may be confusing and may encourage people to install the newest software and as far as I understood the Vitis 2021.x.x has not been integrated yet with  KRS.

giuliocorradi, 2022-01-31:
good suggestion to include in the KRS install instructions..**.if you do not  find the version 20xx.x.x,  look in the archive**

jasvinderkhurana, 2022-09-13:
tools are now updated, will add proper instructions to install.

closed 2022-09-13 by jasvinderkhurana