Xilinx/KRS #40: Include "std_msgs/msg/string.hpp" not found during colcon build

kevinkeryk, 2022-03-09:
Built ROS 2 under Ubuntu 20.04 with Vivado 2021.2 tools installed following these instructions:
http://docs.ros.org/en/rolling/Installation/Ubuntu-Development-Setup.html

Talker and listener examples are working but now when I follow the beta install directions for KRS:
https://github.com/vmayoral/KRS/blob/beta/sphinx/source/docs/install.md

I am running into the following fatal error during the `colcon build --merge-install` step:

```
Finished <<< multiple_doublevadd_publisher [18.2s]
    --- stderr: ros2_kria_power  
In file included from /home/training/krs_ws/src/acceleration/ros2_kria/ros2_kria_power/src/power_component.cpp:25:
/home/training/krs_ws/src/acceleration/ros2_kria/ros2_kria_power/include/ros2_kria_power/power_component.hpp:30:10: fatal error: std_msgs/msg/string.hpp: No such file or directory
  30 | #include "std_msgs/msg/string.hpp"
     |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/power_component.dir/build.make:63: CMakeFiles/power_component.dir/src/power_component.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:136: CMakeFiles/power_component.dir/all] Error 2
gmake: *** [Makefile:141: all] Error 2
Failed   <<< ros2_kria_power [18.4s, exited with code 2]
Aborted  <<< doublevadd_publisher [19.1s]
```

This causes other packages to fail build as well.  Seems like something rather basic is missing from my environment for it to not be able to find string.hpp but I am not sure where to start searching.  Has anyone else run into this issue?

vmayoral, 2022-03-09:
> [](https://github.com/Xilinx/KRS/issues/40#issue-1164106773)Built ROS 2 under Ubuntu 20.04 with Vivado 2021.2 tools installed following these instructions:
http://docs.ros.org/en/rolling/Installation/Ubuntu-Development-Setup.html

That's really not required, you can install ROS from .deb files directly following http://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debians.html. 

[std_msgs](https://github.com/ros/std_msgs) is a very basic package. If basic pub/sub is working for your, it should be operating over `std_msgs`. I can't reproduce this in my machine. Do you think you can build a Dockerfile that demonstrates this issue @kevinkeryk?

vmayoral, 2022-03-10:
@kevinkeryk did you fix your issues installing ROS 2 from deb. files?

kevinkeryk, 2022-03-10:
Yes, I was able to avoid this by installing ROS 2 from .deb files according to these directions instead:

http://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debians.html

closed 2022-03-10 by kevinkeryk