Xilinx/KRS #41: Include "experimental/xrt_bo.h" not found during colcon build

kevinkeryk, 2022-03-09:
Installed ROS 2 packages under Ubuntu 20.04 along with Vitis 2021.2 tools installed following these instructions:
http://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debians.html

Talker and listener examples are working but now when I follow the beta install directions for KRS:
https://github.com/vmayoral/KRS/blob/beta/sphinx/source/docs/install.md

I am running into the following fatal error during the `colcon build --merge-install` step:

```
[Processing: acceleration_firmware_kv260, image_proc]
--- stderr: image_proc                                                                                                          
In file included from /home/training/krs_ws/src/perception/image_pipeline/image_proc/src/rectify_resize_fpga_integrated_xrt.cpp:35:
/home/training/krs_ws/src/perception/image_pipeline/image_proc/include/image_proc/rectify_resize_fpga_integrated_xrt.hpp:37:10: fatal error: experimental/xrt_bo.h: No such file or directory
   37 | #include "experimental/xrt_bo.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rectify_resize_fpga_xrt.dir/build.make:63: CMakeFiles/rectify_resize_fpga_xrt.dir/src/rectify_resize_fpga_integrated_xrt.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:624: CMakeFiles/rectify_resize_fpga_xrt.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< image_proc [2min 37s, exited with code 2]
Aborted  <<< acceleration_firmware_kv260 [5min 25s]

```

This causes other packages to fail build as well.  Seems like perhaps something from XRT is missing from my environment for it to not be able to find xrt_bo.h but I am not sure where to start searching.  Has anyone else run into this issue?

vmayoral, 2022-03-09:
Have you installed XRT @kevinkeryk?

vmayoral, 2022-03-10:
@surender009, @kevinkeryk and @tomasthoresen, reported struggles with this.

#### Problem
In a nutshell, the errors in here are caused by [the following](https://github.com/ros-acceleration/image_pipeline/blob/ros2/image_proc/CMakeLists.txt#L107-L122) (and similar XRT-enabled components):

```cmake
# rectify_resize_fpga_xrt component (integrated, with XRT) - 1 Component, 1 kernel
ament_auto_add_library(rectify_resize_fpga_xrt SHARED src/rectify_resize_fpga_integrated_xrt.cpp)
if (DEFINED CMAKE_SYSROOT)
  target_link_options(rectify_resize_fpga_xrt PRIVATE "LINKER:-lxrt_coreutil")
endif()
target_include_directories(rectify_resize_fpga_xrt PUBLIC
  # $<INSTALL_INTERFACE:${CMAKE_INSTALL_PREFIX}/include/vitis_common>
  $ENV{XILINX_HLS}/common/technology/autopilot
  $ENV{XILINX_HLS}/include
  /home/xilinx/XRT/build/Debug/opt/xilinx/xrt/include  # TODO: replace with ROS 2 header-only library
  /usr/include/xrt  # TODO: replace with ROS 2 header-only library
)
target_link_libraries(rectify_resize_fpga_xrt ${OpenCL_LIBRARY})
target_compile_definitions(rectify_resize_fpga_xrt PRIVATE "COMPOSITION_BUILDING_DLL")
rclcpp_components_register_nodes(rectify_resize_fpga_xrt "image_proc::RectifyResizeNodeFPGAXRT")
set(node_plugins "${node_plugins}image_proc::RectifyResizeNodeFPGAXRT;$<TARGET_FILE:rectify_resize_fpga_xrt>\n")
```

CMake logic is seeking for XRT installed in the workstation but if it fails to find it, launches an error complaining about the headers not being found (like the one above). 

#### Solution 1
XRT is a bit tricky, installing it from Xilinx deb files leads to some issues and you may need to compile it from sources to get full capabilities. An easy patch is to hardcode the XRT header's path for each one of your systems/workstations after successfully installing it.
```cmake
target_include_directories(rectify_resize_fpga_xrt PUBLIC
  # $<INSTALL_INTERFACE:${CMAKE_INSTALL_PREFIX}/include/vitis_common>
  $ENV{XILINX_HLS}/common/technology/autopilot
  $ENV{XILINX_HLS}/include
  <include in here your hardcoded path to the headers>
  /usr/include/xrt  # TODO: replace with ROS 2 header-only library
)
```

#### Solution 2
Another approach is to create a ROS 2 header-only library with XRT headers, which will take additional time. I took this approach in https://github.com/ros-acceleration/vitis_common.

#### Solution 3
The simplest one (from a users' consumption perspective, though certainly requires quite a few changes in the source code) in my opinion is to exclude those components from being built unless XRT is properly enabled, which by default, we should discourage (given that more work is needed on that end).

I've implemented this by using a new CMake variabled called `ROS_XRT` which is set (or unset) by the `ament_vitis` ROS build system extensions. In a nutshell, it lands as follows in ROS 2 packages:

```cmake
if (DEFINED ROS_XRT)
    # rectify_resize_fpga_xrt component (integrated, with XRT) - 1 Component, 1 kernel
    ament_auto_add_library(rectify_resize_fpga_xrt SHARED src/rectify_resize_fpga_integrated_xrt.cpp)
    if (DEFINED CMAKE_SYSROOT)
      target_link_options(rectify_resize_fpga_xrt PRIVATE "LINKER:-lxrt_coreutil")
    endif()
    target_include_directories(rectify_resize_fpga_xrt PUBLIC
      # $<INSTALL_INTERFACE:${CMAKE_INSTALL_PREFIX}/include/vitis_common>
      $ENV{XILINX_HLS}/common/technology/autopilot
      $ENV{XILINX_HLS}/include
      /home/xilinx/XRT/build/Debug/opt/xilinx/xrt/include  # TODO: replace with ROS 2 header-only library
      /usr/include/xrt  # TODO: replace with ROS 2 header-only library
    )
    target_link_libraries(rectify_resize_fpga_xrt ${OpenCL_LIBRARY})
    target_compile_definitions(rectify_resize_fpga_xrt PRIVATE "COMPOSITION_BUILDING_DLL")
    rclcpp_components_register_nodes(rectify_resize_fpga_xrt "image_proc::RectifyResizeNodeFPGAXRT")
    set(node_plugins "${node_plugins}image_proc::RectifyResizeNodeFPGAXRT;$<TARGET_FILE:rectify_resize_fpga_xrt>\n")
  endif()  # ROS_XRT
```

This provides additional syntax to enrich CMakeLists.txt files.

vmayoral, 2022-03-10:
I think the most interesting one (from a usability perspective for evaluating KRS `beta`) is "**Solution 3**", so I'll be focusing on enabling that one via a pull request to KRS with the changes necessary.

vmayoral, 2022-03-10:
@kevinkeryk, @jasvinderkhurana, @surender009, @thomasthoresen, https://github.com/Xilinx/KRS/pull/43 addresses the issues described in here. Please try that out and let me know.

vmayoral, 2022-03-10:
@surender009 just reported suceededing after #43

kevinkeryk, 2022-03-10:
Okay, I cleaned up my workspace and started following this instead to move past this issue:

https://github.com/Xilinx/KRS/blob/main/sphinx/source/docs/install.md