Xilinx/KRS #45: CMake Error when building gazebo_ros

kevinkeryk, 2022-03-10:
Installed ROS 2 packages under Ubuntu 20.04 along with Vitis 2021.2 tools installed following these instructions:
http://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debians.html

Talker and listener examples are working but now when I follow the main install directions for KRS:
https://github.com/Xilinx/KRS/blob/main/sphinx/source/docs/install.md

I am running into the following CMake Error during the `colcon build --merge-install` step:

```
Starting >>> gazebo_ros
--- stderr: gazebo_ros                                                                                                                                   
CMake Error at /home/training/krs_ws/install/share/gazebo_dev/cmake/gazebo_dev-extras.cmake:2 (find_package):
  By not providing "Findgazebo.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gazebo", but
  CMake did not find one.

  Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake

  Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
  "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /home/training/krs_ws/install/share/gazebo_dev/cmake/gazebo_devConfig.cmake:41 (include)
  CMakeLists.txt:26 (find_package)

---
Failed   <<< gazebo_ros [1.74s, exited with code 1]
```

Anyone else seeing this issue?

kevinkeryk, 2022-03-10:
I came across this post here which seems to indicate that Gazebo developer packages need to be installed for this part of the build to succeed:
https://discuss.px4.io/t/error-attempting-to-build-gazebo-on-ubuntu-20-10/20440/2

So I tried installing `libgazebo9-dev` with `sudo apt install libgazebo9-dev`as suggested in the post.

However, this gets me to a point in colcon build where there is an unbelievable amount of errors in some of the header files linked to by the gazebo_plugins package.

Did some further research and found that there is now a `libgazebo11-dev` which is available and can be installed with `sudo apt install libgazebo11-dev` so I uninstalled the `libgazebo9-dev` package with `sudo apt remove libgazebo9-dev` and installed `libgazebo11-dev` instead.

This appears to now allow colcon to finish building all 91 packages including gazebo_plugins (with some warnings, but no errors).

Can we add `libgazebo11-dev` to the dependencies list in the instructions?  Or are there separate steps that should be followed instead for installing Gazebo and its plugins?

vmayoral, 2022-03-10:
Is this while cross-compiling? Or while building it for the workstation?
The latter should work.

These packages shouldn’t be cross-compiled, note the instructions in the
example:

colcon build --build-base=build-kv260 --install-base=install-kv260
--merge-install --mixin kv260 --cmake-args -DNOKERNELS=true
--packages-up-to perception_2nodes image_pipeline_examples

kevinkeryk, 2022-03-10:
I believe that I am building for the workstation since I am just finishing these instructions here and got stuck on the above issue:
https://github.com/Xilinx/KRS/blob/main/sphinx/source/docs/install.md

I am just now moving on to Example 0:
https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/0_ros2_publisher.html

vmayoral, 2022-03-16:
@kevinkeryk have you progressed in here? Have a look at https://gitlab.com/xilinxrobotics/docs/-/issues/57#note_876401297 wherein I reviewed it again for the sake of supporting you. I experienced a minor strugle with some updates in the Rolling packages but addressed it (and explain there how to do so). This is nevertheless a very different problem than the one you're experiencing.

If you don't have access to Gitlab's repo, send me your Gitlab username and I'll enable you.

kevinkeryk, 2022-03-17:
Yes, I was able to move past this issue by installing the `libgazebo11-dev` dependency.  Not sure if it is adding headers or what but it seems to be satisfying that CMake package configuration for gazebo.  If there is a better way to install gazebo that we should be following, please let me know.

closed 2022-03-17 by kevinkeryk