Xilinx/KRS #46: #hello_xilinx example in an emulated KV260 hangs at ATF

surender009, 2022-03-16:
surender@xhdsurender41:/scratch/krs_beta$ *colcon acceleration emulation sw_emu install-kv260*
SECURITY WARNING: This class invokes explicitly a shell via the shell=True argument of the Python subprocess library, and uses admin privileges to manage raw disk images. It is the user's responsibility to ensure that all whitespace and metacharacters passed are quoted appropriately to avoid shell injection vulnerabilities.
- Verified that install/ is available in the current ROS 2 workspace
- Confirmed availability of raw image file at: /scratch/krs_beta/acceleration/firmware/select/sd_card.img
- Finished inspecting raw image, obtained UNITS and STARTSECTOR P1/P2
[sudo] password for surender:
- Image mounted successfully at: /tmp/sdcard_img_p2
- Successfully cleaned up prior overlay ROS 2 workspace at: /tmp/sdcard_img_p2/krs_beta
- Copied 'install-kv260' directory as a ROS 2 overlay workspace in the raw image at location: /krs_beta.
- Created and copied in rootfs /opt/ros/foxy/setup.bash.
- Umounted the raw image.
- Generated PMU and QEMU files.
- Launching emulation...
cd /scratch/krs_beta/acceleration/firmware/select/emulation && /proj/xbuilds/SWIP/2021.2_1021_0703/installs/lin64/Vitis/2021.2/bin/launch_emulator -device-family ultrascale -target sw_emu -qemu-args-file /scratch/krs_beta/acceleration/firmware/select/emulation/qemu_args.txt -pmc-args-file /scratch/krs_beta/acceleration/firmware/select/emulation/pmu_args.txt -sd-card-image /scratch/krs_beta/acceleration/firmware/select/sd_card.img -enable-prep-target $*
CRITICAL_WARNING: [LAUNCH_EMULATOR] DEPRECATED !! Using the old flow which uses launch_emulator.tcl. Please use v++ -p to generate the script to launch new launch_emulator.py
Running SW Emulation
INFO :  [LAUNCH_EMULATOR] pl_sim_dir option is not provided.
pmc_dtb /proj/xbuilds/SWIP/2021.2_1021_0703/installs/lin64/Vitis/2021.2/data/emulation/dtbs/zynqmp/zynqmp-pmu.dtb
resized sd card image
Starting QEMU
 - Press <Ctrl-a h> for help
Waiting for QEMU to start. qemu_port 8965
Qemu_pids 170767 170768
qemu-system-aarch64: -chardev socket,path=./qemu-rport-_pmu@0,server,id=pmu-apu-rp: info: QEMU waiting for connection on: disconnected:unix:./qemu-rport-_pmu@0,server
QEMU started. qemu_pid=170767.
Waiting for PMU to start. Qemu_pids 170771 170772
qemu-system-aarch64: -chardev socket,id=pl-rp,host=127.0.0.1,port=8160,server: info: QEMU waiting for connection on: disconnected:tcp:127.0.0.1:8160,server
PMC started. pmc_pid=170771
Starting PL simulation.Generating PLLauncher commandline
running PLL Launcher
PMU Firmware 2020.2     Jun 10 2021   22:45:10
PMU_ROM Version: xpbr-v8.1.0-0
NOTICE:  BL31: v2.4(release):xlnx_rebase_v2.4_2021.1_update1-23-g9188496b9
NOTICE:  BL31: Built : 07:41:24, Oct 13 2021

vmayoral, 2022-03-16:
Hardware and software emulation were never worked in Vitis 2021.2 BSPs so emulation's not part of the KRS `beta` capabilities see https://github.com/ros-acceleration/acceleration_firmware_kv260.

surender009, 2022-03-16:
ok Thanks @vmayoral . i am closing this

closed 2022-03-16 by surender009