Xilinx/KRS #47: #Hello_Xilinx example building failure on zcu102 firmware

surender009, 2022-03-16:
## i followed below instructions from KRS beta and am seeing build errors for **hello_xilinx** example targeting zcu102
[hello_xilinx on zcu102 emulation](https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/1_hello_xilinx.html)

--- stderr: publisher_xilinx
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/proj/xbuilds/SWIP/2021.2_1021_0703/installs/lin64/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /scratch/krs_beta/build-zcu102/publisher_xilinx/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/make cmTC_3a2c1/fast && /usr/bin/make -f CMakeFiles/cmTC_3a2c1.dir/build.make CMakeFiles/cmTC_3a2c1.dir/build
    make[1]: Entering directory '/scratch/krs_beta/build-zcu102/publisher_xilinx/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_3a2c1.dir/testCCompiler.c.o
    /proj/xbuilds/SWIP/2021.2_1021_0703/installs/lin64/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-gcc --sysroot=/scratch/krs_beta/install/../acceleration/firmware/zcu102/sysroots/aarch64-xilinx-linux    -o CMakeFiles/cmTC_3a2c1.dir/testCCompiler.c.o   -c /scratch/krs_beta/build-zcu102/publisher_xilinx/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_3a2c1
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_3a2c1.dir/link.txt --verbose=1
    /proj/xbuilds/SWIP/2021.2_1021_0703/installs/lin64/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-gcc --sysroot=/scratch/krs_beta/install/../acceleration/firmware/zcu102/sysroots/aarch64-xilinx-linux      CMakeFiles/cmTC_3a2c1.dir/testCCompiler.c.o  -o cmTC_3a2c1
    /proj/xbuilds/SWIP/2021.2_1021_0703/installs/lin64/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-ld.real: cannot find crtbeginS.o: No such file or directory
    /proj/xbuilds/SWIP/2021.2_1021_0703/installs/lin64/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-ld.real: cannot find -lgcc
    collect2.real: error: ld returned 1 exit status
    make[1]: *** [CMakeFiles/cmTC_3a2c1.dir/build.make:87: cmTC_3a2c1] Error 1
    make[1]: Leaving directory '/scratch/krs_beta/build-zcu102/publisher_xilinx/CMakeFiles/CMakeTmp'
    make: *** [Makefile:121: cmTC_3a2c1/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

---
Failed   <<< publisher_xilinx [3.39s, exited with code 1]

Summary: 0 packages finished [3.71s]
  1 package failed: publisher_xilinx
  1 package had stderr output: publisher_xilinx

vmayoral, 2022-03-16:
Refer to KRS `Alpha` for instructions that are known to work. There's no `Beta` support for ZCU102. If you compare https://github.com/ros-acceleration/acceleration_firmware_zcu102 with https://github.com/ros-acceleration/acceleration_firmware_kv260, you'll not this aspect and that there's no beta release for the ZCU102 

As an additional bit, ZCU102 is not anymore a supported platform @surender009 for KRS AFAIK. This was decided early during development to reduce the targets so that I could advance faster in terms of depth (functionality) as opposed to spend time providing support to other hardware and BSPs.

closed 2022-03-16 by surender009

vmayoral, 2022-03-16:
Anytime @surender009!

vmayoral, 2022-03-16:
Actually @surender009, I think there's merit in your ask in the sense that it wasn't as clear as it could be, so I've gone ahead and pushed https://github.com/ros-acceleration/acceleration_firmware_zcu102/commit/01e9215ade71a1a39f43aa9c26a63d2e3fcf2917. Now it's coherent with KV260's firmware as well.

I leave it to you to consider adding a note to the KRS Documentation pointing readers to alpha for the ZCU102.