Xilinx/KRS #49: Physical I/O interaction with constrain file (*.xdc)

teerameth, 2022-03-18:
I have already made myself through examples to create accelerated ROS2 node in this repo but I haven't seen any detail about interfacing to physical pin which usually determined by *.xdc files in Vivado. Did I miss something or I have to do some additional steps ?
I plane to change from normal HLS workflow to KRS in my senoir thesis, if there is a tutorial about interfacing to physical hardware it will be perfect !
Thank in advance.

vmayoral, 2022-03-18:
@teerameth can you elaborate? What exactly are you looking for in connection with ROS? Note KRS focuses on the ROS integration, but the default (and traditional) Vitis and Vivado flows remain usable, as always.

Can you provide a draft example of what you’d like to see and what such a “ROS example” will achieve? Be specific about hardware interfacing please.

teerameth, 2022-03-18:
I'm looking for something like "Node that read digital input from some digital pin then publish as ROS topic" or "ROS2 service which recieve I2C address then scanning via digital pins (as SCL, SDA) and return whoami register of avaliable devices".
Thankyou.