Xilinx/KRS #50: Issue with source the ROS 2 overlay workspace - 0 ROS 2 publisher

mohammedrafi-sk, 2022-03-19:
While trying the [vadd_publisher](https://github.com/ros-acceleration/acceleration_examples/tree/main/nodes/vadd_publisher) example onboard, the step to source the ROS 2 overlay workspace ` .  /ros2_ws/local_setup.bash,` is failing as there is no ros2_ws folder in `/`

```
xilinx-k26-som-2021_2:~$ ls -l /
total 64
drwxr-xr-x   2 root root  4096 Mar 19 03:11 bin
drwxr-xr-x   3 root root  4096 Mar 19 03:11 boot
drwxr-xr-x   4 root root     0 Dec 31  1969 configfs
drwxr-xr-x  13 root root 14540 Mar  9  2018 dev
drwxr-xr-x  58 root root  4096 Mar 19 04:12 etc
drwxr-xr-x   4 root root  4096 Mar 19 03:11 home
lrwxrwxrwx   1 root root    10 Mar 19 03:11 init -> /sbin/init
drwxr-xr-x   4 root root  4096 Mar 19 03:11 krs_ws
drwxr-xr-x   9 root root  4096 Mar 19 03:11 lib
drwx------   2 root root 16384 Mar 19 03:11 lost+found
drwxr-xr-x   4 root root  4096 Dec 31  1969 media
drwxr-xr-x   2 root root  4096 Dec 31  1969 mnt
drwxr-xr-x   3 root root  4096 Mar 19 03:11 opt
dr-xr-xr-x 163 root root     0 Dec 31  1969 proc
drwxr-xr-x   7 root root   620 Mar  9  2018 run
drwxr-xr-x   3 root root  4096 Mar 19 03:11 sbin
dr-xr-xr-x  13 root root     0 Dec 31  1969 sys
lrwxrwxrwx   1 root root     8 Mar  9  2018 tmp -> /var/tmp
drwxr-xr-x  11 root root  4096 Mar 19 03:11 usr
drwxr-xr-x   9 root root  4096 Mar  9  2018 var
```

I believe `. /krs_ws/local_setup.bash`  is the corrent one

vmayoral, 2022-03-19:
No, you are mixing things up.

krs_ws is the workspace in your workstation to build KRS. ros2_ws is the resulting cross-compiled install directory that you need to scp into the embedded board.

It’s explained as such in the documentation. Refer to it.

mohammedrafi-sk, 2022-03-19:
In the [double_vadd](https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/0_ros2_publisher.html?highlight=ros2_ws#doublevadd-publisher) it mentiones we can scp the content to board.  But in the [vadd-publisher](https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/0_ros2_publisher.html?highlight=ros2_ws#vadd-publisher), it says the content is already copied to sd_card.img

```
$ . /ros2_ws/local_setup.bash     # source the ROS 2 overlay workspace we just
                                  # created. Note it has been copied to the SD
                                  # card image while being created.
```

vmayoral, 2022-03-19:
> But in the [vadd-publisher](https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/0_ros2_publisher.html?highlight=ros2_ws#vadd-publisher), it says the content is already copied to sd_card.img.

Correct, the following line does it:

> colcon acceleration linux vanilla --install-dir install-kv260

Which can be traced back to:
- https://github.com/ros-acceleration/colcon-acceleration/blob/main/colcon_acceleration/subverb/linux.py#L505
- https://github.com/ros-acceleration/colcon-acceleration/blob/19829c2155d0469cd5d8ae15b2e0a92bf32fe99d/colcon_acceleration/subverb/__init__.py#L645

I recall having enforced the naming `ros2_ws` somewhere but don't quite find it right now. @mohammedrafi-sk can you check if you got a `/install-kv260` instead in your embedded device? If so, that's the actual confusion and it'd need to be fixed in the documentation (and a PR would be more than welcome!).

Renaming or scp-ing the folder will do for you.

mohammedrafi-sk, 2022-03-19:
No, I don't see any `/install-kv260` folder on the board. 
Here is the content on the root folder https://github.com/Xilinx/KRS/issues/50#issue-1174243382

vmayoral, 2022-03-19:
> No, I don't see any /install-kv260 folder on the board.

@mohammedrafi-sk, see the documentation please. And **follow** that. Particularly:

> You can do this in various ways (physically mounting the raw image in the SD card, scp-ing the ROS 2 workspace, etc.).

As pointed out [above](https://github.com/Xilinx/KRS/issues/50#issuecomment-1073023651), depending on how you create your SD card raw image the cross-compiled workspace might (or might not) be copied for you but regardless, you can just drop it yourself in embedded. It's a simple ROS 2 overlay workspace. Source it and make use of its content.

mohammedrafi-sk, 2022-03-19:
Okay, After copying the content with scp , I am able to run the publisher. 
Thanks