Xilinx/KRS #51: Issue with generating bit while building simple adder

mohammedrafi-sk, 2022-03-19:
While building the [simple_adder](https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/2_hls_ros2.html#simple-adder1-a-quick-look-into-the-flow), I am seeing the below error.

```
--- stderr: simple_adder
ERROR: The following input file does not exist: /home/mr/krs_ws/build-kv260/simple_adder/simple_adder.xclbin
[ERROR]  : Cannot read BIT file - simple_adder.bit
---
Finished <<< simple_adder [1min 49s]
```

I figured the error was due to the unavailability of board part xilinx.com:k26c:part0:1.1 in Vivado 2021.2 as shown in the log [v++_simple_adder.log](https://github.com/Xilinx/KRS/files/8309591/v%2B%2B_simple_adder.log),  I downloaded all the available file revisions for this board from the catalogue, but I only see 1.2 and 1.3 but no 1.1

Further, If I rebuild, the error goes away, but there was no xclbin/bit in `~/krs_ws/build-kv260/simple_adder`. 
But, if I do rebuild by adding  `--cmake-clean-cache` flag to force CMake configure, I see the error every time. 
`colcon build  --merge-install --build-base=build-kv260 --install-base=install-kv260 --mixin kv260 --packages-select simple_adder --cmake-clean-cache`

I am seeing the same issue with offloaded_doublevadd too.

vmayoral, 2022-03-21:
Hello @mohammedrafi-sk, this is known to work but I think some of the Xilinx/AMD folks should be able to help you debugging this in your setup and see if there's some additional input to incorporate in the documentation.

Friendly ping to @kevinkeryk, @jasvinderkhurana, @surender009 and @giuliocorradi.

jasvinderkhurana, 2022-09-13:
@mohammedrafi-sk, were you able to resolve this, if not let work together on this.

sachinkum0009, 2024-01-01:
Hi, 

I am also facing this issue related to `Cannot read BIT file` for simple_adder as well as offloaded_double_vadd while building.

Can you please help to solve this issue?

Thanks

sachinkum0009, 2024-01-04:
@vmayoral @jasvinderkhurana 

Can you please help to solve this issue?

Thanks 🙏

jasvinderkhurana, 2024-01-05:
@sachinkum0009,

Can you please let me know if you are using Ubuntu flow or petalinux flow? I will check this at my end and will let you know.

sachinkum0009, 2024-01-06:
Thanks, I am using Petalinux flow with KV260.

vmayoral, 2024-01-06:
Hello folks,

I haven't reproduced this myself and without further investigation I can only make a few guesses. 

----

Shortly, It's highly like that folks are missing some dependencies, drivers (or devices in Vitis/Vivado) versions, as hinted by @mohammedrafi-sk. This is likely causing the Vivado build effort to fail, and thereby ending up in the errors you see reported above (which I fully agree, could be more descriptive however we didn't have time to improve things on this regard in the past).

----

@mohammedrafi-sk your report at https://github.com/Xilinx/KRS/issues/51#issue-1174303698 is pretty good. It'd be useful to add to that additional details (Vitis/Vivado suite being used? 2021.2?) for reproducibility so that we could give it a try if the issues persists. Shortly, from what I've understood, several of you are experiencing:

```bash
--- stderr: simple_adder
ERROR: The following input file does not exist: /home/mr/krs_ws/build-kv260/simple_adder/simple_adder.xclbin
[ERROR]  : Cannot read BIT file - simple_adder.bit
---
Finished <<< simple_adder [1min 49s]
```

I'll break this a bit further and explain it:

```
[ERROR]  : Cannot read BIT file - simple_adder.bit
```

This error is reporting that the `simple_adder.bit` file wasn't found in the build process. This is likely caused because Vivado is encountering an error. Such an error can be found in the Vivado log files generated in the corresponding build stage directory: `/home/mr/krs_ws/build-kv260/simple_adder/`. The way to debug this out would be to search in the vivado files generated under that directory and find the error.

```
ERROR: The following input file does not exist: /home/mr/krs_ws/build-kv260/simple_adder/simple_adder.xclbin
```

This error line is related to the one above. Since the `.bit` file wasn't generated, the `.xclbin` file (which is a container for the `.bit`) is neither generated and thereby, things similarly fail.

Bottom line,

> Look for the error found in the Vivado log files generated in the corresponding build stage directory: `/home/mr/krs_ws/build-kv260/simple_adder/`.

Forward looking, the error handling can certainly improve. given the complexity of the layers integrated in here (ROS 2 build system (`ament`), ROS 2 build tools (`colcon`), Vitis, Vivado and plenty of Tcl logic) it's probably a good idea to consider reporting better errors (e.g. outputing in stdout the errors from Vivado, instead of keeping them in log files). This would however require additional development.

sachinkum0009, 2024-01-06:
Thanks for the suggestions. I am using Vivado 2022.1

This is the error message in log

```bash
[3.511s] ERROR: [v++ 82-29] Exception in main() -- ERROR: [v++ 84-18] Application not found: 'vitis_hls'.
[3.511s] 
[3.511s] INFO: [v++ 60-1662] Stopping dispatch session having empty uuid.
[3.511s] INFO: [v++ 60-1653] Closing dispatch client.
[3.525s] -- CMD_ERROR: 1
````

It is quite weird, as the vitis_hls is available through cli. 

```bash
>>echo $XILINX_VIVADO 
/media/asus/backup/Xilinx/Vivado/2022.1
>>echo $XILINX_VITIS 
/media/asus/backup/Xilinx/Vitis/2022.1
>>echo $XILINX_HLS 
/media/asus/backup/Xilinx/Vitis_HLS/2022.1
```

I am attaching the log file.

[log file](https://github.com/Xilinx/KRS/files/13849502/build_2024-01-06_12-47-33.zip)

vmayoral, 2024-01-06:
> ```shell
> [3.511s] ERROR: [v++ 82-29] Exception in main() -- ERROR: [v++ 84-18] Application not found: 'vitis_hls'.
> [3.511s] 
> [3.511s] INFO: [v++ 60-1662] Stopping dispatch session having empty uuid.
> [3.511s] INFO: [v++ 60-1653] Closing dispatch client.
> [3.525s] -- CMD_ERROR: 1
> ```

Seems to be a path issue then in the corresponding environmental variables for the `ament` (CMake) build process. In principle. Last time I tested, the KRS tooling dealt with that.

sachinkum0009, 2024-01-06:
Thanks for the response. TBH I am not much sure about the issue, any suggestion on how to solve it.

sachinkum0009, 2024-01-15:
Hi, 
Can you please help with this issue?
Is it more related to installation of Vivado? Should I try different version?

Thanks

ACESUSUSU, 2024-11-26:
Hi,I am also facing this issue related to Cannot read BIT file for simple_adder as well as offloaded_double_vadd while building.
And for most situation,its just because the mem is not enough,after I MAKE SWAP bigger, it can be build successfully.
refer to https://fpga.eetrend.com/blog/2021/100553175.html