Xilinx/KRS #53: ROS 2 build-farm integration of KRS core packages

vmayoral, 2022-03-29:
This ticket follows from https://github.com/ros2-gbp/ros2-gbp-github-org/pull/22#issuecomment-1073811491 and tracks the progress of releasing into the official ROS 2 buildfarm generic hardware acceleration-related KRS ROS packages.

ping @methylDragon, luca-della-vedova. Please me keep this updated.

#### Hardware acceleration packages release status
- [ ] `package 1`: [colcon-acceleration](https://github.com/ros-acceleration/colcon-acceleration)
  - [x] evaluation 
    - https://github.com/ros-acceleration/colcon-acceleration/issues/10
    - https://github.com/colcon/colcon-core/issues/493#issuecomment-1071350845
  - [x] code fixes (assigned to: @vmayoral)
    - [x] `colcon` team to review package name and general approach https://github.com/colcon/colcon-core/issues/493#issuecomment-1071350845
    - [x] Resolve their concerns
    - [ ] (*optional*) recommendations https://github.com/ros-acceleration/colcon-acceleration/issues/10#issue-1168063785 @methylDragon
  - [ ] release
    - [ ] Create [`publish-python.yaml`](https://github.com/dirk-thomas/publish-python) file to define the release process
        - See example in [`colcon-core`](https://github.com/colcon/colcon-core/blob/master/publish-python.yaml)
    - [x] Raise issue in [`colcon/colcon-core`](https://github.com/colcon/colcon-core) to request for this repo to be transferred into the `colcon` org (https://github.com/colcon/colcon-core/issues/493)
      - [ ] Migrate package
    - [ ] PR for implementation in the repo in that org to be sent and approved if necessary
- [ ] `package 2`: [ament_acceleration](https://github.com/ros-acceleration/ament_acceleration)
  - [x] evaluation https://github.com/ros-acceleration/ament_acceleration/issues/1
  - [x] code fixes (assigned to: @vmayoral)
    - [x] Create a **release tag** corresponding to the version specified in `package.xml` @vmayoral 
    - https://github.com/ros-acceleration/ament_acceleration/commit/f1568b1a0edc8055c04ca230c523d7f01b653cdc
  - [ ] release
    - [ ] Create release repo (pending https://github.com/ros2-gbp/ros2-gbp-github-org/pull/22)
    - [ ] Do [`bloom` release](https://wiki.ros.org/bloom/Tutorials/FirstTimeRelease)
      - [ ] `bloom` should PR `rosdistro`, if not, **manually create a PR**
    - [ ] `rosdistro` team to review package and merge PR (1-2 days, 1 month to get apt installable)
    - [ ] Monitor build and release status on [repo page](http://repo.ros2.org/status_page/) and [buildfarm](https://build.ros2.org/)
- [ ] `package 3`: [ament_vitis](https://github.com/ros-acceleration/ament_vitis)
  - [x] evaluation https://github.com/ros-acceleration/ament_vitis/issues/1
  - [x] code fixes (assigned to: @vmayoral)
    - [x] Create a **release tag** corresponding to the version specified in `package.xml` @vmayoral 
    - https://github.com/ros-acceleration/ament_vitis/commit/d9ba9c3f342d982fc9d8da816868c15bbdadfa66
    - https://github.com/ros-acceleration/ament_vitis/commit/48683473415eb8a3a2f6e1bd6c4eace635357100
    - https://github.com/ros-acceleration/ament_vitis/commit/bb72b0a8d9b403ebdbc2755708b414674a333c11
  - [ ] release
    - [ ] Create release repo (pending https://github.com/ros2-gbp/ros2-gbp-github-org/pull/22)
    - [ ] Do [`bloom` release](https://wiki.ros.org/bloom/Tutorials/FirstTimeRelease)
      - [ ] `bloom` should PR `rosdistro`, if not, **manually create a PR**
    - [ ] `rosdistro` team to review package and merge PR (1-2 days, 1 month to get apt installable)
    - [ ] Monitor build and release status on [repo page](http://repo.ros2.org/status_page/) and [buildfarm](https://build.ros2.org/)
- [ ] `package 4`: [vitis_common](https://github.com/ros-acceleration/vitis_common)
  - [x] evaluation https://github.com/ros-acceleration/vitis_common/issues/1
  - [x] code fixes (assigned to: @vmayoral)
    - [x] Create a **release tag** corresponding to the version specified in `package.xml` @vmayoral 
  - [ ] release
    - [ ] Create release repo (pending https://github.com/ros2-gbp/ros2-gbp-github-org/pull/22)
    - [ ] Do [`bloom` release](https://wiki.ros.org/bloom/Tutorials/FirstTimeRelease)
      - [ ] `bloom` should PR `rosdistro`, if not, **manually create a PR**
    - [ ] `rosdistro` team to review package and merge PR (1-2 days, 1 month to get apt installable)
    - [ ] Monitor build and release status on [repo page](http://repo.ros2.org/status_page/) and [buildfarm](https://build.ros2.org/)
- [ ] ~~`package 5`: [acceleration_firmware](https://github.com/ros-acceleration/acceleration_firmware)~~
  - [x] ~~evaluation and discussion https://github.com/ros-acceleration/acceleration_firmware/issues/1~~
  - [ ] ~~code fixes ⚫ (assigned to: TBD), I couldn't come up with any good scenario wherein (other than for organizational matters) keeping this package will add significant value from a source code perspective so I'm <ins>biased to recommend to drop it completely</ins>, and modify other packages accordingly. The effort of doing so (modify other packages) is something manageable if considering future firmware releases. See https://github.com/ros-acceleration/acceleration_firmware/issues/1#issuecomment-1070687656~~.
  - [ ] ~~release~~
- [ ] `package 6`: [tracetools_acceleration](https://github.com/ros-acceleration/tracetools_acceleration)
  - [x] evaluation https://github.com/ros-acceleration/tracetools_acceleration/issues/1
  - [x] code fixes (assigned to: @vmayoral)
    - [x] Create a **release tag** corresponding to the version specified in `package.xml` @vmayoral 
  - [ ] release
    - [ ] Create release repo (pending https://github.com/ros2-gbp/ros2-gbp-github-org/pull/22)
    - [ ] Do [`bloom` release](https://wiki.ros.org/bloom/Tutorials/FirstTimeRelease)
      - [ ] `bloom` should PR `rosdistro`, if not, **manually create a PR**
    - [ ] `rosdistro` team to review package and merge PR (1-2 days, 1 month to get apt installable)
    - [ ] Monitor build and release status on [repo page](http://repo.ros2.org/status_page/) and [buildfarm](https://build.ros2.org/)
- [ ] `package 7`: [ros2acceleration](https://github.com/ros-acceleration/ros2acceleration)
  - [x] evaluation https://github.com/ros-acceleration/ros2acceleration/issues/2
  - [x] code fixes (assigned to: @vmayoral)
    - [x] Create a **release tag** corresponding to the version specified in `package.xml` @vmayoral 
  - [ ] release
    - [ ] Create release repo (pending https://github.com/ros2-gbp/ros2-gbp-github-org/pull/22)
    - [ ] Do [`bloom` release](https://wiki.ros.org/bloom/Tutorials/FirstTimeRelease)
      - [ ] `bloom` should PR `rosdistro`, if not, **manually create a PR**
    - [ ] `rosdistro` team to review package and merge PR (1-2 days, 1 month to get apt installable)
    - [ ] Monitor build and release status on [repo page](http://repo.ros2.org/status_page/) and [buildfarm](https://build.ros2.org/)
#### Additional feedback
- https://github.com/colcon/colcon-core/issues/493#issuecomment-1071350845 (🟠)
- https://github.com/ros-acceleration/KRS_private/issues/1 (🟠)
- [Releasing_ROS_2_Packages.pdf](https://github.com/ros2-gbp/ros2-gbp-github-org/files/8315206/Releasing_ROS_2_Packages.pdf)

####  ROS packages release process
1. Getting the package sources in working order (including creating a release tag on the source repository)
2. Making a release repository (separate from the source repository)
3. Doing a release, pushing it to the release repository
4. Sending a PR to rosdistro and getting it merged
5. Checking the buildfarm for successful builds, and waiting for the sources to get synced to the official ROS debian repositories

####  Additional tickets to track
- https://github.com/ros2-gbp/ros2-gbp-github-org/pull/24

vmayoral, 2022-03-29:
@methylDragon, @luca-della-vedova I brought this information in here to continue tracking progress.

Reviewed the progress and added the dependency between `ament_acceleration` and `ament_vitis`. Let me know if you have any additional input!

vmayoral, 2022-03-29:
@methylDragon, on https://github.com/ros-acceleration/tracetools_acceleration/issues/1#issuecomment-1081891662 a PR would be appreciated.

vmayoral, 2022-03-30:
Fulfilled for all repos http://wiki.ros.org/buildfarm/Pull%20request%20testing

methylDragon, 2022-03-30:
Nice!! I'll be doing the releases enabling PR testing moving forward then

vmayoral, 2022-04-05:
Reviewed package status and updated things accordingly above.

Looking further into `colcon-hardware-acceleration`.

vmayoral, 2022-04-06:
All right, I think we're getting there. For the sake of alignment, thinks that I can track and are left:
- `vitis_common` WIP https://github.com/ros/rosdistro/pull/32671
- `ament_acceleration` pending PRs seem merged https://github.com/ros-acceleration/ament_acceleration/issues/1#issuecomment-1086201266, are we waiting for something else in here? Can we ask for re-trigering the build maybe? If so, is there an easy way for me to do it?
- colcon-hardware-acceleration has been pushed to https://pypi.org/project/colcon-hardware-acceleration/ but there're two new tags. The latest one at https://github.com/colcon/colcon-hardware-acceleration/pull/15. **Important**, I don't have privileges to merge this PR nor to push directly to this repo after the migration. Can we address all these so that we can submit 0.7.0 to PyPI?

Looking at [Humble Freeze dates](https://docs.ros.org/en/rolling/Releases/Release-Humble-Hawksbill.html#release-timeline) @methylDragon, anything that concerns you at this stage to make it *safely*?

methylDragon, 2022-04-06:
We have until May 19 and most of the packages are already merged into `rosdistro`. There shouldn't be any issues unless `vitis_common` has binary build failures. Syncs will happen and don't require us to fit the package in after the deadline (the deadline just means the packages have to be frozen.)

The release for `adaptive_component` will be a bit tighter though, especially if we want to make time to make sure it's really robust.. But in principle if we just want a release, it can follow the same sort of process as per the packages we've released so far.

vmayoral, 2022-04-07:
All right, it appears only `package 1` is left to be completed with the overall KRS core packages release.

@methylDragon do we have any updates on?

> - colcon-hardware-acceleration has been pushed to https://pypi.org/project/colcon-hardware-acceleration/ but there're two new tags. The latest one at [...]

methylDragon, 2022-04-07:
No updates here, unfortunately.. The `colcon` team doesn't have capacity assigned to this, so any support we get from them is from review time, that I've been trying to push for. This is for both PyPI and packagecloud.

The bizdev side people might be better to ask for progress on getting that maneuvered around, but we should be able to meet the humble freeze (also, the package is already available on PyPI anyway.)

But in the meantime I'll ping the `colcon` team to see if we can be added as maintainers.

---

What I'm more concerned about is whether we'll be able to complete Task 2 by the time. Would it be possible to start looking into those issues? I haven't been able to fully test the boards/run through the testing tutorials yet.

vmayoral, 2022-04-07:
> No updates here, unfortunately.. The colcon team doesn't have capacity assigned to this, so any support we get from them is from review time, that I've been trying to push for. This is for both PyPI and packagecloud.

Ok thanks. 

To get clarity on my side and report back, does this mean that (both for now and future contributions) the bottleneck of [`colcon-hardware-acceleration`](https://github.com/colcon/colcon-hardware-acceleration) contributions and PR reviews will be the colcon team? If so, and no access/privileges are given to us after transferring the repo, doesn't this mean that effectively, this repo will be maintained by the colcon team? 

Maybe I misunderstood something in here, but this wasn't intended.

> What I'm more concerned about is whether we'll be able to get the acceleration nodes merged by the time. Would it be possible to start looking into those issues? I haven't been able to fully test the boards/run through the testing tutorials yet.

What do you mean by "*to get the acceleration nodes merged by the time*"? The instructions that I have is for me to work with you guys to meet the objectives above (`milestone 1`) before anything else. Happy to support you while trying to run the acceleration examples if you're encountering problems. If so, could you open tickets with the corresponding issues (e.g. within [acceleration_examples](https://github.com/ros-acceleration/acceleration_examples))?

*Note that these examples aren't meant to be released in Humble/Rolling just yet*.

methylDragon, 2022-04-07:
I'm trying to get us added as maintainers, I think it'll be better for both the colcon team and the hardware-acceleration-wg in that case.

But for now there are no new actionables aside from waiting for colcon, and for the sync and making sure the currently released packages get propagated to the public repos.

---

Let's start working on the hardware acceleration nodes in the meantime.

There's a couple of information gathering issues asking about the intention currently and moving forward for certain decisions made in the implementation of `adaptive_component`. Should we track this in a separate issue?

- https://github.com/ros-acceleration/KRS_private/issues/3
- https://github.com/ros-acceleration/adaptive_component/issues/3
- https://github.com/ros-acceleration/adaptive_component/issues/2

methylDragon, 2022-04-07:
Ok, both of us should be added as maintainers on the colcon org's hwa team!

@vmayoral , would you like to try doing the PyPI release?

<details>
<summary>How to do it</summary>

With the publish-python binary and folder in the colcon package, run
```
# This targets the PyPI release and allows you to skip the packagecloud one!
./bin/publish-python -upload wheel:pypi
```
</details>

vmayoral, 2022-04-08:
Thanks @methylDragon, uploaded `0.7.0` and confirmed I have privileges in both the repo and PyPI.

AFAIK, this wraps all items above for releasing key KRS repos. A few open questions:
- is there anything left unfinished for the "release process" of the packages above?
- are we good with the build-farm regarding all the packages above? I can see some *missing* aspects in red with most of them (e.g. http://repo.ros2.org/status_page/ros_rolling_default.html?q=vitis_common) 
- can you clarify the situation with packagecloud and `colcon-hardware-acceleration`? Should we be doing releases in here as well?

methylDragon, 2022-04-08:
Hi @vmayoral ,

- The packages listed above will need to be bloomed again for `humble` once `humble` is released, but aside from that there is nothing for us to do except to wait!
- I think we're good on the build farm also, the red squares for all of them mean that they haven't been pushed and made available to the apt repositories yet (so you can't `apt install` them). This will need to wait for a [sync](https://discourse.ros.org/t/preparing-for-rolling-sync-2022-03-24/24521), and those tend to come around once a month.
  - Once the sync happens, those red squares should turn blue or green. We just need to make sure that they're included in the sync!!

As for packagecloud... The colcon team is trying to get the permissions to add us as maintainers to the colcon org's packagecloud. I think you can give them some time, and once they get those permissions, you can run this command to push to it instead (making sure to edit `publish-python.yaml` to point to the colcon org's packagecloud repo:
`./bin/publish-python -upload stdeb:packagecloud`

If they can't get the permissions, I think they'll look for a workaround. I'm not sure about the timeline for this.

vmayoral, 2022-04-08:
All right, thanks. That clarifies a lot. I'll annotate then the checkboxes above to reflect pending things accordingly.

vmayoral, 2022-04-13:
I can see @methylDragon that we're full-green now (e.g. http://repo.ros2.org/status_page/ros_rolling_default.html?q=ament_acceleration) but I did a quick check searching in Open Robotics ROS 2 apt repository and couldn't find these in Rolling (`ros-rolling-*`).

Maybe my expectation was wrong and there're some additional steps needed in the process until these become ready?

methylDragon, 2022-04-13:
We need to wait for the sync to complete. It takes about a week or so, but our packages are in already.

See the [new package list](https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2022-04-12/25092)

If you can, confirm all the packages we've released are there.

methylDragon, 2022-04-13:
I see some references in the source list: 
[image]

methylDragon, 2022-04-13:
[image]
[image]

methylDragon, 2022-04-13:
http://packages.ros.org/ros2/ubuntu/lists/

methylDragon, 2022-04-13:
Oh, also, rolling package releases on Linux only target Jammy if I'm not wrong. Try it out on a Jammy machine?

I our packages in:
- http://packages.ros.org/ros2/ubuntu/lists/ros-rolling-jammy-amd64_jammy_main_amd64_Packages
- http://packages.ros.org/ros2/ubuntu/lists/ros-rolling-jammy-arm64_jammy_main_arm64_Packages
- http://packages.ros.org/ros2/ubuntu/lists/ros-rolling-jammy-source_jammy_main_Sources

But not in (expected):
- http://packages.ros.org/ros2/ubuntu/lists/ros-rolling-jammy-i386_jammy_main_i386_Packages
- http://packages.ros.org/ros2/ubuntu/lists/ros-rolling-jammy-armhf_jammy_main_armhf_Packages

methylDragon, 2022-04-21:
`adaptive_component` should be in `rolling` after [this sync](https://discourse.ros.org/t/preparing-for-rolling-sync-2022-04-20/25167)

methylDragon, 2022-04-23:
https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2022-04-22/25213 It's in!

methylDragon, 2022-04-29:
Colcon package is up on http://repo.ros2.org/status_page/rolling_default.html and http://repo.ros2.org/status_page/humble_default.html !!

methylDragon, 2022-04-29:
Now we're only waiting for a humble sync and we should be done!

vmayoral, 2022-05-16:
Thanks everyone for the effort in here!