Xilinx/KRS #55: Finding that colcon build fails without Vitis tools being installed

kevinkeryk, 2022-04-12:
I want to validate that colcon can build the dependencies for simulation to run on the workstation without the need for Vitis tools (think of a user going through the Out-of-Box exercise to see the connected target board running our pre-built binaries) and I am following these instructions for install here:

https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html

I am encountering this error during the colcon build step when I do not source the `/tools/Xilinx/Vitis/2021.2/settings64.sh` script:

```
--- stderr: image_proc
In file included from /home/training/krs_ws/install/include/vitis_common/common/xf_common.hpp:20,
                 from /home/training/krs_ws/install/include/vitis_common/common/xf_sw_utils.hpp:20,
                 from /home/training/krs_ws/install/include/vitis_common/common/xf_headers.hpp:28,
                 from /home/training/krs_ws/src/perception/image_pipeline/image_proc/src/resize_fpga.cpp:32:
/home/training/krs_ws/install/include/vitis_common/common/xf_structs.hpp:27:10: fatal error: ap_axi_sdata.h: No such file or directory
   27 | #include "ap_axi_sdata.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/resize_fpga.dir/build.make:63: CMakeFiles/resize_fpga.dir/src/resize_fpga.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:420: CMakeFiles/resize_fpga.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< image_proc [1.01s, exited with code 2]
```

Steps to recreate this error under Ubuntu 20.04 LTS workstation:

1. Open a new Terminal
2. Change to KRS workspace with `cd ~/krs_ws`
3. Remove workspace build and install folders `rm -rf build install` to eliminate intermediate results
4. Source ROS 2 installation `source /opt/ros/rolling/setup.bash` to setup environment for building
5. Adjust path for Cmake `export PATH="/usr/bin":$PATH` to call the correct Cmake version
6. Run colcon build  `colcon build --merge-install` to start building dependency packages

It is my understanding that the host components can be built without any Vitis dependencies.  Anyone else seeing a similar issue?

vmayoral, 2022-04-13:
A super quick fix is to obviously use the colcon flags to ignore the failing package since it's not needed for the flow you're trying to enable but I believe we should probably get to the root of this. Spent the whole morning looking at this, haven't been able to find the issue so far. Will continue looking at it this afternoon.

vmayoral, 2022-04-13:
All right, after quite a few hours, tracked down the error. Essentially the `beta` branch in `ament_vitis` doesn't include fixes to detect board-specific that I've recently [introduced](https://github.com/ros-acceleration/ament_vitis/blob/main/cmake/ros_vitis.cmake#L21-L25). Accordingly, right now, the CMake logic is not stopping the workstation from entering in the `ROS_VITIS` which leads to the error you're observing since that's not getting built.

Fixes for this issue are available today in `main` branches of various of the packages involved (ament_vitis, vitis_common, image_pipeline).

Once the next release is out, this problem should dissapear but I won't be making another beta release since that's very costly.

kevinkeryk, 2022-04-13:
Thank you for checking into this @vmayoral and I had a follow-up question on this.

Should I simply update these lines in my `krs_rolling.repos` before I build with colcon if I wish to continue testing with `beta` otherwise?

```
  acceleration/ament_vitis:
    type: git
    url: https://github.com/ros-acceleration/ament_vitis
    version: main
```

I am following the `beta` instructions here:  https://github.com/Xilinx/KRS/blob/main/sphinx/source/docs/install.md

vmayoral, 2022-04-14:
No, that’ll clash with other dependencies. It’ll require a whole (chained)
release of packages.

Let me think if I can have the code ignored for the workstation flow solely
using CMake logic, BUT it’ll be hack only for you. This has been already
properly fixed in main branches.

vmayoral, 2022-04-14:
All right, so I spent a while again looking at this and created https://github.com/ros-acceleration/image_pipeline/tree/beta2, which addresses your direct problem above. The issue though (as indicated above), is that as I was fearing, this is triggered other packages to crash with dumps like the following one:

```bash
Finished <<< perception_2nodes [9.80s]
--- stderr: image_pipeline_examples
WARNING: XILINX_VITIS is not defined, hardware acceleration disabled.
WARNING: XILINX_VIVADO is not defined, hardware acceleration disabled.
WARNING: XILINX_HLS is not defined, hardware acceleration disabled.
/usr/bin/ld: CMakeFiles/rectify_resize_fpga_integrated_streamlined_node.dir/src/rectify_resize_fpga_integrated_streamlined_node.cpp.o: in function `void __gnu_cxx::new_allocator<image_proc::RectifyResizeNodeFPGAStreamlined>::construct<image_proc::RectifyResizeNodeFPGAStreamlined, rclcpp::NodeOptions>(image_proc::RectifyResizeNodeFPGAStreamlined*, rclcpp::NodeOptions&&)':
/usr/include/c++/9/ext/new_allocator.h:146: undefined reference to `image_proc::RectifyResizeNodeFPGAStreamlined::RectifyResizeNodeFPGAStreamlined(rclcpp::NodeOptions const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rectify_resize_fpga_integrated_streamlined_node.dir/build.make:257: rectify_resize_fpga_integrated_streamlined_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:248: CMakeFiles/rectify_resize_fpga_integrated_streamlined_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: CMakeFiles/rectify_resize_fpga_integrated_node.dir/src/rectify_resize_fpga_integrated_node.cpp.o: in function `void __gnu_cxx::new_allocator<image_proc::RectifyResizeNodeFPGA>::construct<image_proc::RectifyResizeNodeFPGA, rclcpp::NodeOptions>(image_proc::RectifyResizeNodeFPGA*, rclcpp::NodeOptions&&)':
/usr/include/c++/9/ext/new_allocator.h:146: undefined reference to `image_proc::RectifyResizeNodeFPGA::RectifyResizeNodeFPGA(rclcpp::NodeOptions const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rectify_resize_fpga_integrated_node.dir/build.make:257: rectify_resize_fpga_integrated_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:167: CMakeFiles/rectify_resize_fpga_integrated_node.dir/all] Error 2
/usr/bin/ld: CMakeFiles/rectify_resize_fpga_node.dir/src/rectify_resize_fpga_node.cpp.o: in function `void __gnu_cxx::new_allocator<image_proc::RectifyNodeFPGA>::construct<image_proc::RectifyNodeFPGA, rclcpp::NodeOptions&>(image_proc::RectifyNodeFPGA*, rclcpp::NodeOptions&)':
/usr/include/c++/9/ext/new_allocator.h:146: undefined reference to `image_proc::RectifyNodeFPGA::RectifyNodeFPGA(rclcpp::NodeOptions const&)'
/usr/bin/ld: CMakeFiles/rectify_resize_fpga_node.dir/src/rectify_resize_fpga_node.cpp.o: in function `void __gnu_cxx::new_allocator<image_proc::ResizeNodeFPGA>::construct<image_proc::ResizeNodeFPGA, rclcpp::NodeOptions&>(image_proc::ResizeNodeFPGA*, rclcpp::NodeOptions&)':
/usr/include/c++/9/ext/new_allocator.h:146: undefined reference to `image_proc::ResizeNodeFPGA::ResizeNodeFPGA(rclcpp::NodeOptions const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rectify_resize_fpga_node.dir/build.make:257: rectify_resize_fpga_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:140: CMakeFiles/rectify_resize_fpga_node.dir/all] Error 2
/usr/bin/ld: CMakeFiles/rectify_resize_fpga_streamlined_node.dir/src/rectify_resize_fpga_streamlined_node.cpp.o: in function `void __gnu_cxx::new_allocator<image_proc::RectifyNodeFPGAStreamlined>::construct<image_proc::RectifyNodeFPGAStreamlined, rclcpp::NodeOptions&>(image_proc::RectifyNodeFPGAStreamlined*, rclcpp::NodeOptions&)':
/usr/include/c++/9/ext/new_allocator.h:146: undefined reference to `image_proc::RectifyNodeFPGAStreamlined::RectifyNodeFPGAStreamlined(rclcpp::NodeOptions const&)'
/usr/bin/ld: CMakeFiles/rectify_resize_fpga_streamlined_node.dir/src/rectify_resize_fpga_streamlined_node.cpp.o: in function `void __gnu_cxx::new_allocator<image_proc::ResizeNodeFPGAStreamlined>::construct<image_proc::ResizeNodeFPGAStreamlined, rclcpp::NodeOptions&>(image_proc::ResizeNodeFPGAStreamlined*, rclcpp::NodeOptions&)':
/usr/include/c++/9/ext/new_allocator.h:146: undefined reference to `image_proc::ResizeNodeFPGAStreamlined::ResizeNodeFPGAStreamlined(rclcpp::NodeOptions const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rectify_resize_fpga_streamlined_node.dir/build.make:257: rectify_resize_fpga_streamlined_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:194: CMakeFiles/rectify_resize_fpga_streamlined_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< image_pipeline_examples [21.7s, exited with code 2]
Aborted  <<< image_proc_adaptive [22.8s]
Aborted  <<< gazebo_ros [1min 30s]

Summary: 86 packages finished [3min 51s]
  1 package failed: image_pipeline_examples
  2 packages aborted: gazebo_ros image_proc_adaptive
  50 packages had stderr output: accelerated_doublevadd_publisher ament_clang_format ament_clang_tidy ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_lint ament_lint_cmake ament_mypy ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_uncrustify ament_xmllint camera_calibration colcon-hardware-acceleration depth_image_proc doublevadd_publisher faster_doublevadd_publisher image_pipeline_examples image_proc image_proc_adaptive image_rotate launch launch_pytest launch_testing launch_xml launch_yaml multiple_doublevadd_publisher offloaded_doublevadd_publ [... cut ...]
As a final note, @kevinkeryk, ⚠️ note these are **only hot-fixes to support your use case (`which implies changing the instructions of the documentation`) but a proper fix has already been addressed in main branches**. I have not fully tested things functionality-wise, only ensured it builds⚠️. 

We will need to wait until I make the next release. For a fully tested complete release (again coherent to documentation, if the flow of the documentation changes (as in this case), of course adjustments will be required).

jasvinderkhurana, 2022-04-19:
@vmayoral, is it possible to get a patch before we do a next release? This will help users avoid Vitis dependency in execution of Perception stack application.

vmayoral, 2022-04-25:
> @vmayoral, is it possible to get a patch before we do a next release? This will help users avoid Vitis dependency in execution of Perception stack application.

@jasvinderkhurana Refer to my previous comment with `beta2` for you guys:

<details><summary>krs_rolling_beta2.repos</summary>

```bash
cat << 'EOF' > krs_rolling_beta2.repos
repositories:
  ros2/ament_lint:
    type: git
    url: https://github.com/ament/ament_lint
    version: master
  ros2/launch:
    type: git
    url: https://github.com/ros2/launch
    version: master
  ros2/gazebo_ros_pkgs:
    type: git
    url: https://github.com/ros-simulation/gazebo_ros_pkgs
    version: da14a69e79502cd08064ccd261366ff023a9162a
  
  perception/image_common:
    type: git
    url: https://github.com/ros-perception/image_common
    version: 9729de81f7dff6156f644d6152b200f687360f1f
  perception/image_pipeline:
    type: git
    url: https://github.com/ros-acceleration/image_pipeline
    version: beta2
  perception/vision_opencv:
    type: git
    url: https://github.com/ros-perception/vision_opencv
    version: 7bbc5ecc232e8faf36b45efaa2b6bc979b04157f

  tracing/ros2_tracing:
    type: git
    url: https://gitlab.com/ros-tracing/ros2_tracing.git
    version: master
  tracing/tracetools_acceleration:
    type: git
    url: https://github.com/ros-acceleration/tracetools_acceleration
    version: main

  acceleration/acceleration_firmware:
    type: git
    url: https://github.com/ros-acceleration/acceleration_firmware
    version: main
  acceleration/acceleration_firmware_kv260:
    type: zip
    url: https://github.com/ros-acceleration/acceleration_firmware_kv260/releases/download/v0.9.0/acceleration_firmware_kv260.zip
  acceleration/adaptive_component:
    type: git
    url: https://github.com/ros-acceleration/adaptive_component
    version: main
  acceleration/ament_acceleration:
    type: git
    url: https://github.com/ros-acceleration/ament_acceleration
    version: main
  acceleration/ament_vitis:
    type: git
    url: https://github.com/ros-acceleration/ament_vitis
    version: beta
  acceleration/colcon-acceleration:
    type: git
    url: https://github.com/ros-acceleration/colcon-acceleration
    version: main
  acceleration/ros2_kria:
    type: git
    url: https://github.com/ros-acceleration/ros2_kria
    version: main
  acceleration/ros2acceleration:
    type: git
    url: https://github.com/ros-acceleration/ros2acceleration
    version: main
  acceleration/vitis_common:
    type: git
    url: https://github.com/ros-acceleration/vitis_common
    version: master
  acceleration/acceleration_examples:
    type: git
    url: https://github.com/ros-acceleration/acceleration_examples
    version: beta2

EOF
```

</details>

jasvinderkhurana, 2022-06-29:
@kevinkeryk , I assume this problem should be resolved in latest release, So closing the ticket

closed 2022-06-29 by jasvinderkhurana