Xilinx/KRS #58: ament_cuda not found

mohammedrafi-sk, 2022-06-29:
This error is seen while building the KRS1.0 for kr260

```
CMake Warning at CMakeLists.txt:13 (find_package):
  By not providing "Findament_cuda.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cuda", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cuda" with
  any of the following names:

    ament_cudaConfig.cmake
    ament_cuda-config.cmake

  Add the installation prefix of "ament_cuda" to CMAKE_PREFIX_PATH or set
  "ament_cuda_DIR" to a directory containing one of the above files.  If
  "ament_cuda" provides a separate development package or SDK, be sure it has
  been installed.
```

Is this expected? Did I miss some dependencies on the host?

mohammedrafi-sk, 2022-06-29:
As a workaround, I cloned the[ ament_cuda](https://github.com/ros-acceleration/ament_cuda) to src/accelerator and then the build finished with no issues

jasvinderkhurana, 2022-06-29:
@mohammedrafi-sk , Ideally user should not clone cuda part as this must be for Nvidia. Let me also check on this and will see if this is a bug.

vmayoral, 2022-06-29:
This is not an error, it's a warning. It's set up that way to enable multi-vendor support in *related* packages (it's what's used to generate benchmarks across compute substrates).

Cloning `ament_cuda` is not necessary at all. Warning can also be removed if necessary but I don't see any issue with it as is.

mohammedrafi-sk, 2022-06-29:
Oh yes, Its a warning,  
Thanks for clarifying.