Xilinx/KRS #59: Failed to cross-compile CPU binaries - kr260

mohammedrafi-sk, 2022-06-29:
The following errors are seen while building the cross-compile CPU binaries for kr260 
[KRS1.0_kr260_crosscompile_cpu.txt](https://github.com/Xilinx/KRS/files/9008824/KRS1.0_kr260_crosscompile_cpu.txt)

### 7.A cross-compile and generate ONLY CPU binaries
###################################################
colcon build --build-base=build-kr260-ubuntu --install-base=install-kr260-ubuntu --merge-install --mixin kr260 --cmake-args -DNOKERNELS=true

mohammedrafi-sk, 2022-06-29:
For resolving the error with module "ament_cmake_core", I cross-compiled the ament_cmake, 
```
Could not find a package configuration file provided by "ament_cmake_core"
  with any of the following names:
```

Now, I am seeing similar issue but with ament_cmake_ros
```
Failed   <<< tracetools_image_pipeline [0.73s, exited with code 1]
--- stderr: tracetools_acceleration
CMake Error at CMakeLists.txt:15 (find_package):
  By not providing "Findament_cmake_ros.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_ros", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake_ros"
  with any of the following names:

    ament_cmake_rosConfig.cmake
    ament_cmake_ros-config.cmake

  Add the installation prefix of "ament_cmake_ros" to CMAKE_PREFIX_PATH or
  set "ament_cmake_ros_DIR" to a directory containing one of the above files.
  If "ament_cmake_ros" provides a separate development package or SDK, be
  sure it has been installed.

---
Failed   <<< tracetools_acceleration [0.74s, exited with code 1]

```

vmayoral, 2022-06-29:
ament_cmake_core should be installed in the Ubuntu sysroot if deployed appropriately. Check it it’s there please and otherwise report what went wrong in the deployment process (and posterior patching).

Also, to help assess the issues, it’d be useful to get the whole commands with their corresponding errors, as well as instructions on how to reproduce the errors. Otherwise providing help is hard.

mohammedrafi-sk, 2022-06-29:
Hi @vmayoral 

I can't seem to find any ROS related CMake files in the ubuntu sysroot ``` ~/krs_ws/acceleration/firmware/kr260/sysroots  ``` .
```
spa@ros2server:~/krs_ws/acceleration/firmware/kr260/sysroots$ find . -name ament_*
spa@ros2server:~/krs_ws/acceleration/firmware/kr260/sysroots$ 
```

I found this patch which seems to be installing ROS onto sysroot, can you confirm?
``` ~/krs_ws/acceleration/firmware/kr260/patches/customize.patch ```

mohammedrafi-sk, 2022-07-04:
Hi @vmayoral 

I tried rebuilding the KRS as you mentioned, and found the chroot failed and so the ros packages didn't install to sysroot
Here is the log

```
mr@krs:~/krs_ws$ colcon build --merge-install
Starting >>> ament_acceleration
Starting >>> tracetools_image_pipeline
Starting >>> tracetools_acceleration
Starting >>> adaptive_component
Starting >>> camera_calibration
Starting >>> depth_image_proc
Starting >>> image_publisher
Starting >>> image_rotate
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' insteadtion - 0.2s] [depth_image_proc - 0.2s] [image_publisher - 0.2s] [image_rotate - 0.2s]
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
  warnings.warn(
--- stderr: camera_calibration
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' instead
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< camera_calibration [6.90s]
Starting >>> image_view
Finished <<< ament_acceleration [7.49s]
Starting >>> ament_vitis
Finished <<< tracetools_image_pipeline [7.36s]
Starting >>> ros2_kria_msgs
Finished <<< tracetools_acceleration [7.38s]
Starting >>> FortranCInterface
Finished <<< ament_vitis [1.58s]
Starting >>> vitis_common
Finished <<< ros2_kria_msgs [1.55s]
Starting >>> IntelFortranImplicit
[23.011s] WARNING:colcon.colcon_cmake.task.cmake.build:Could not run installation step for package 'IntelFortranImplicit' because it has no 'install' target
Finished <<< IntelFortranImplicit [1.26s]
Starting >>> acceleration_firmware_kr260
Finished <<< vitis_common [1.36s]
Starting >>> image_proc
[23.331s] WARNING:colcon.colcon_cmake.task.cmake.build:Could not run installation step for package 'FortranCInterface' because it has no 'install' target
Finished <<< FortranCInterface [3.11s]
Starting >>> colcon-hardware-acceleration
--- stderr: colcon-hardware-acceleration
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< colcon-hardware-acceleration [1.00s]
Starting >>> publisher_xilinx
[Processing: acceleration_firmware_kr260, adaptive_component, depth_image_proc, image_proc, image_publisher, image_rotate, image_view, publisher_xilinx]
[Processing: acceleration_firmware_kr260, adaptive_component, depth_image_proc, image_proc, image_publisher, image_rotate, image_view, publisher_xilinx]
--- stderr: acceleration_firmware_kr260
+ set -o pipefail
+ trap 'finish "$chroot_d"' EXIT
+ '[' 5 -lt 4 ']'
+ '[' 0 -ne 0 ']'
+ chroot_d=/home/mr/krs_ws/install/../acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux
+ suite=jammy
+ ppa_src=http://packages.ros.org/ros2/ubuntu
+ ppa_token=
++ dirname /home/mr/krs_ws/install/../acceleration/firmware/kr260/sysroots/bin/customize.sh
+ scripts_d=/home/mr/krs_ws/install/../acceleration/firmware/kr260/sysroots/bin
+ shift 4
+ pkgs_in_private_archive=(xrt linux-headers-xilinx-zynqmp)
+ cp /home/mr/krs_ws/install/../acceleration/firmware/kr260/sysroots/bin/cleanup-chroot.sh /home/mr/krs_ws/install/../acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux
+ cd /home/mr/krs_ws/install/../acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux
[... 41 lines cut ...]
+ read -r submount
+ umount /home/mr/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/proc
+ read -r submount
+ umount /home/mr/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/dev/shm
+ read -r submount
+ umount /home/mr/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/dev/pts
+ read -r submount
+ umount /home/mr/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/dev/mqueue
+ read -r submount
+ umount /home/mr/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/dev/hugepages
+ read -r submount
+ umount /home/mr/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/dev
+ read -r submount
---
Finished <<< acceleration_firmware_kr260 [2min 6s]

```

But, after installing qemu on the Ubuntu host, the chroot worked, and I don't see any of the above ros package dependency issues.
```
sudo apt install qemu-user-static 
```

I have the vitis installed, but qemu is not part of it. I believe qemu is part of petalinux and not Vitis
@jasvinderkhurana Can you confirm if qemu is also part of Vitis installation?

elimkwan, 2022-08-24:
Hi @vmayoral @mohammedrafi-sk,

I have encountered the same chroot error when trying to cross-compile the CPU binaries for kr260. 
```
+ chroot . apt update
chroot: failed to run command ‘apt’: Exec format error
```
Unfortunately, the fix `sudo apt install qemu-user-static` did not work for me. Hence, I would like to reopen the issue. 

So far I have not been able to find a fix, except to compute the CPU binaries on target. I have tried it on a Windows machine with wsl2 and a Linux machine (Ubuntu 22.04), and both of them result in the same chroot error, so it doesn't seem to be a wsl2 issue. Any insight would be appreciated, thanks.

verderog, 2022-10-28:
This thread has been very helpful!  I'm building KRS using Docker and it turns out that when launching the container, I did not enable the `--privileged` flag.  This in turn caused the...

```
/krs_ws/install/../acceleration/firmware/kr260/sysroots/bin/customize.sh
```
...script to silently fail during the `prepare_chroot()` function.  This failure caused errors relating to `ament_cmake_core` and similar `ament_cmake*` packages when executing:

```
colcon build --build-base=build-kr260-ubuntu --install-base=install-kr260-ubuntu --merge-install --mixin kr260 --cmake-args -DNOKERNELS=true
```

Posting this in case anyone else runs into this in the future.

lanyangyang, 2022-11-18:
@verderog Could you please explain more clearly about how you fix this issue ?? I met the same issue as you mentioned.

verderog, 2022-11-22:
> @verderog Could you please explain more clearly about how you fix this issue ?? I met the same issue as you mentioned.

For context, I've created a Docker container in order to build KRS.  This Docker container performs the operations that the KRS build instructions call out (see more [here](https://github.com/Xilinx/KRS/blob/main/sphinx/source/docs/install.md)).  

It turns out that within a Docker container running the `prepare_chroot()` function that is within the `customize.sh` script will fail, due to some of the `mount` operations that are performed.  By launching the container with the `--privileged` flag, that causes the `prepare_chroot()` to execute successfully.  If the function fails (because the `mount` operations can't complete), the side effect is that you run into those various `ament_cmake*` errors.

More info on the Docker privileged flag [here](https://docs.docker.com/engine/reference/commandline/run/#full-container-capabilities---privileged).  One of the key take-aways: 

> ...most potentially dangerous kernel capabilities are dropped; including cap_sys_admin (which is required to mount filesystems). However, the --privileged flag will allow it to run...

I hope this helps!

vmayoral, 2022-11-22:
> It turns out that within a Docker container running the `prepare_chroot()` function that is within the `customize.sh` script will fail, due to some of the `mount` operations that are performed. By [...]

Right, this requires sudo privileges to successfully install ROS 2 in the chroot. The reason why some people is experiencing failures is because these privileges are not being given appropriately (either inside or outside of the container). By adding the `--privileged` flag (and by default executing with root privileges inside) you're addressing this.

Outside of the container folks should pay attention to this (and/or `sudo -s` before `colcon build --merge-install`ing). There's a user experience gap in here that I agree needs to be addressed.