Xilinx/KRS #62: Error while building accelerator for KR260 - device tree overlay blob generation

vmayoral, 2022-07-02:
I caught a weird error while re-running things in a clean workspace with the KR260:

**Description**: The [KR260 KRS v1.0.0 firmware](https://github.com/ros-acceleration/acceleration_firmware_kr260/releases/tag/v1.0.0) is built minimalistic and ships as little as possible. While building accelerators, the CMake logic attempts to produce all necessary artifacts (besides the accelerator itself) including the `.dtbo` file and the `shell.json` files necessary to run things on top of `libdfx`. This can be tracked down to https://github.com/ros-acceleration/ament_vitis/blob/main/cmake/vitis_acceleration_kernel.cmake#L357-L372. As things are, the accelerator will be successfully built but the CMake logic will fail because it won't find necessary files to complete the process.

To <ins>reproduce this issue</ins>, in a clean ROS 2 workspace with KR260 as explained at https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#cross-compilation-development:

```bash
colcon build --build-base=build-kr260-ubuntu --install-base=install-kr260-ubuntu --merge-install --mixin kr260 --cmake-args -DNOKERNELS=false --packages-select simple_vadd  # or any other accelerator
```

vmayoral, 2022-07-02:
Updated comment above with a workaround that should unblock those testing things out. For a complete fix, refer to https://github.com/ros-acceleration/acceleration_firmware_kr260/commit/af069d8e9696168555a26e2ecee3baf28a75330c. This needs to be further validated and fully tested but once ready, it should be [released](https://github.com/ros-acceleration/acceleration_firmware_kr260/releases), and that should do.

vmayoral, 2022-07-02:
Updated again with a more detailed walkthrough on how to apply the workaround.

mohammedrafi-sk, 2022-07-12:
Hi @vmayoral 

Thanks for the workaround. I will clean up and follow it. 

Meanwhile, I tested the complete fix and I am seeing the below error while the building accelerator for KR260

```
[Processing: offloaded_doublevadd_publisher]                                     
[Processing: offloaded_doublevadd_publisher]                                       
--- stderr: offloaded_doublevadd_publisher                                            
cp: cannot stat '/home/mr/krs_ws/install-kr260-ubuntu/../acceleration/firmware/select/data/../device_tree/kernel_default.dtbo': No such file or directory
cp: cannot stat '/home/mr/krs_ws/install-kr260-ubuntu/../acceleration/firmware/select/data/../shell.json': No such file or directory
ln: failed to create symbolic link '/home/mr/krs_ws/install-kr260-ubuntu/../acceleration/firmware/select/data/../BOOT.BIN': No such file or directory
In file included from /home/mr/krs_ws/install-kr260-ubuntu/include/vitis_common/common/ros_opencl_120.hpp:41,
                 from /home/mr/krs_ws/src/acceleration/acceleration_examples/nodes/offloaded_doublevadd_publisher/src/offloaded_doublevadd_publisher.cpp:24:
/home/mr/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/include/CL/cl2.hpp:18:151: note: ‘#pragma message: cl2.hpp has been renamed to opencl.hpp to make it clear that it supports all versions of OpenCL. Please include opencl.hpp directly.’
   18 | #pragma message("cl2.hpp has been renamed to opencl.hpp to make it clear that it supports all versions of OpenCL. Please include opencl.hpp directly.")
      |                                                                                                                                                       ^
CMake Error at cmake_install.cmake:46 (file):
  file INSTALL cannot find
  "/home/mr/krs_ws/build-kr260-ubuntu/offloaded_doublevadd_publisher/vadd_offloaded.dtbo":
  No such file or directory.

---
Failed   <<< offloaded_doublevadd_publisher [53min 15s, exited with code 1]
```

colcon seems to be looking into data folder which doesn't exist in the acceleration firmware. I created a empty data directory in the specified location and then the build worked fine.

I also observed that the platform is missing in kr260 firmware. I used the platform from kv260 firmware for this build.

vmayoral, 2022-07-17:
@mohammedrafi-sk I don't seem to be able to reproduce the error you described after another attempt, moreover the original intent of this ticket was to reflect the workaround needed with KR260 and Ubuntu 22.04 accelerators.

Objective was fulfilled and confirmed (triple-checked) at https://github.com/Xilinx/KRS/issues/67. Closing.

closed 2022-07-17 by vmayoral