Xilinx/KRS #63: camera_info_manager.hpp: No such file or directory

mohammedrafi-sk, 2022-07-04:
The below error is seen while cross-compiling ROS workspace for [ubuntu:kr260](https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#ubuntu-22-04)  Step 5 - ``` colcon build --merge-install ```  

```
mr@krs:~/krs_ws$ colcon build --merge-install
Starting >>> ament_acceleration
Starting >>> tracetools_image_pipeline
Starting >>> tracetools_acceleration
Starting >>> adaptive_component
Starting >>> camera_calibration
Starting >>> depth_image_proc
Starting >>> image_publisher
Starting >>> image_rotate
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' insteadtion - 0.4s] [depth_image_proc - 0.4s] [image_publisher - 0.4s] [image_rotate - 0.4s]
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
  warnings.warn(
Finished <<< ament_acceleration [2.25s]
Starting >>> ament_vitis
Finished <<< tracetools_image_pipeline [2.23s]
Starting >>> image_view
Finished <<< tracetools_acceleration [2.24s]
Starting >>> ros2_kria_msgs
--- stderr: camera_calibration
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' instead
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< camera_calibration [2.86s]
Starting >>> acceleration_firmware_kr260
Finished <<< ros2_kria_msgs [0.85s]
Starting >>> colcon-hardware-acceleration
Finished <<< ament_vitis [0.91s]
Starting >>> vitis_common
--- stderr: image_publisher
CMake Warning at CMakeLists.txt:13 (find_package):
  By not providing "Findament_cuda.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cuda", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cuda" with
  any of the following names:

    ament_cudaConfig.cmake
    ament_cuda-config.cmake

  Add the installation prefix of "ament_cuda" to CMAKE_PREFIX_PATH or set
  "ament_cuda_DIR" to a directory containing one of the above files.  If
  "ament_cuda" provides a separate development package or SDK, be sure it has
  been installed.

/home/mr/krs_ws/src/perception/image_pipeline/image_publisher/src/image_publisher.cpp:34:10: fatal error: camera_info_manager/camera_info_manager.hpp: No such file or directory
   34 | #include <camera_info_manager/camera_info_manager.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/image_publisher.dir/build.make:76: CMakeFiles/image_publisher.dir/src/image_publisher.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/image_publisher.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< image_publisher [3.17s, exited with code 2]
Aborted  <<< colcon-hardware-acceleration [0.30s]
Aborted  <<< acceleration_firmware_kr260 [0.54s]
Aborted  <<< image_rotate [3.47s]
Aborted  <<< depth_image_proc [3.47s]
Aborted  <<< vitis_common [0.49s]
Aborted  <<< image_view [1.42s]
Aborted  <<< adaptive_component [5.60s]

Summary: 6 packages finished [6.64s]
  1 package failed: image_publisher
  7 packages aborted: acceleration_firmware_kr260 adaptive_component colcon-hardware-acceleration depth_image_proc image_rotate image_view vitis_common
  2 packages had stderr output: camera_calibration image_publisher
  19 packages not processed

```

mohammedrafi-sk, 2022-07-04:
Resolved after installing   ros-humble-camera-info-manager ros-humble-camera-info-manager-dbgsym on the host
```
sudo apt install ros-humble-camera-info-manager*
```

vmayoral, 2022-07-17:
`camera_info_manager` is part of https://github.com/ros-perception/image_common, which isn't listed at https://github.com/Xilinx/KRS/blob/main/krs_humble.repos. 

*In principle*, this package should be installed ([instructions at the top](https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html)) if you install Humble Desktop fully, including the image processing packages (e.g. through `ros-humble-desktop-full` target), but this might be wrong and I might've overlooked this. 

I think adding this to the install instructions won't hurt (I don't think `ros-humble-camera-info-manager-dbgsym` is needed though).

jasvinderkhurana, 2022-08-02:
Just make sure to clean the workspace before rebuilding i.e.

rm -rf build install