Xilinx/KRS #65: SOC file not found while cross-compiling ONLY CPU binaries for KR260

mohammedrafi-sk, 2022-07-11:
Got this error while running step [7.4: Cross-compile and generate ONLY CPU binaries](https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#ubuntu-22-04) for generating CPU binaries for KR260

Command:
```
mr@tars:~/krs_ws$ colcon build --build-base=build-kr260-ubuntu --install-base=install-kr260-ubuntu --merge-install --mixin kr260 --cmake-args -DNOKERNELS=true

```

Error:
```
Finished <<< ros2acceleration [5.87s]
Starting >>> simple_adder
--- stderr: simple_adder                                                                                                                                                              
CMake Error at /home/mr/krs_ws/install-kr260-ubuntu/share/ament_vitis/cmake/vitis_hls.cmake:110 (file):
  file STRINGS file
  "/home/mr/krs_ws/install-kr260-ubuntu/../acceleration/firmware/select/SOC"
  cannot be read.
Call Stack (most recent call first):
  CMakeLists.txt:21 (vitis_hls_generate_tcl)

CMake Error at /home/mr/krs_ws/install-kr260-ubuntu/share/ament_vitis/cmake/vitis_hls.cmake:110 (file):
  file STRINGS file
  "/home/mr/krs_ws/install-kr260-ubuntu/../acceleration/firmware/select/SOC"
  cannot be read.
Call Stack (most recent call first):
  CMakeLists.txt:76 (vitis_hls_generate_tcl)

---
Failed   <<< simple_adder [3.87s, exited with code 1]
Aborted  <<< depth_image_proc [5min 46s]                                                                                                                               
Aborted  <<< image_view [5min 24s]                                                                                           
Aborted  <<< image_proc [7min 56s]                                                               

Summary: 14 packages finished [8min 6s]
  1 package failed: simple_adder
  3 packages aborted: depth_image_proc image_proc image_view
  9 packages had stderr output: camera_calibration colcon-hardware-acceleration depth_image_proc image_proc image_publisher image_rotate ros2acceleration simple_adder vitis_common
  15 packages not processed
```

complete log [SOC file not found.txt](https://github.com/Xilinx/KRS/files/9083594/SOC.file.not.found.txt)

vmayoral, 2022-07-11:
Good one @mohammedrafi-sk. This error happens because `/home/mr/krs_ws/install-kr260-ubuntu/../acceleration/firmware/select/SOC` doesn't exist in your development machine and is generally caused by the fact that you have not "selected" the embedded/SOC target which you can do as follows:

```bash
# for the KV260 
colcon acceleration select kv260

# or for the KR260 
colcon acceleration select kr260
```

Technically, this is documented in the examples (e.g. [this one](https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/3_offloading_ros2_publisher.html)) however, with the recent changes in documentations (e.g. with [Ubuntu's enablement](https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#ubuntu-22-04)) I agree this deserves further attention. Maybe a small PR simply adding this command would be helpful @mohammedrafi-sk?

----

Having said that all the above, ⚠️ please see [workaround in here](https://github.com/Xilinx/KRS/issues/62#issue-1292007180) as discussed previously ⚠️, note the limitations explained there about the current firmware. Following those steps will simplify your life. Feel free to contribute with PRs based on your experiences please.

Applepi, 2022-12-05:
I'm also running into this issue building for KR260.

```
CMake Error at /home/nota/krs_ws/install-kr260-ubuntu/share/ament_vitis/cmake/vitis_hls.cmake:110 (file):
  file STRINGS file
  "/home/nota/krs_ws/install-kr260-ubuntu/../acceleration/firmware/select/SOC"
  cannot be read.
Call Stack (most recent call first):
  CMakeLists.txt:21 (vitis_hls_generate_tcl)

CMake Error at /home/nota/krs_ws/install-kr260-ubuntu/share/ament_vitis/cmake/vitis_hls.cmake:110 (file):
  file STRINGS file
  "/home/nota/krs_ws/install-kr260-ubuntu/../acceleration/firmware/select/SOC"
  cannot be read.
Call Stack (most recent call first):
  CMakeLists.txt:76 (vitis_hls_generate_tcl)
  ```
  
It's not clear to me how to fix this. Also documentation about where and how things need to be run as root for Ubuntu 22.04 need serious work. 

I've had issues installing this in a VM so I go to baremetal and it doesn't fix it. I run everything from `source` onward as `sudo -s` to get anything to build. But now after rebuilding like 6 times and nothing is fixing this build error and setting the acceleration does noting at all for kr260. The guide says it should prompt for sudo password it never does as a normal user.

The point of an install guide is that you should be able to run the commands to install the tools without any ambiguity but the entirety of the KRS for cross-compilation and Xilinx tools installation in general are ambiguous. 

"Oh just install 2022.1" but actually you need to install some Ubuntu 22.04 packages before it will install because it will get to the very end of the 66gb download and make you reinstall it again and it re-downloads everything... 

There needs to be some real documentation on how to setup the environment beyond a single line blurb and the constant assumptions made in the install guide.