Xilinx/KRS #66: KR260 Perception node: ros2 acceleration list only shows image_proc

mohammedrafi-sk, 2022-07-12:
I followed the build for perception targeting kr260 as mentioned [here](https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/6_perception.html#building-the-acceleration-kernels)

While testing on board, ``` ros2 acceleration list ``` only shows image_proc. 
````
root@kria:~# ros2 acceleration list
                        Accelerator          Accel_type                            Base           Base_type      #slots(PL+AIE)         Active_slot

accelerated_doublevadd_publisher            XRT_FLATaccelerated_doublevadd_publisher            XRT_FLAT               (0+0)                  -1
                      image_proc            XRT_FLAT                      image_proc            XRT_FLAT               (0+0)                  -1
                k26-starter-kits            XRT_FLAT                k26-starter-kits            XRT_FLAT               (0+0)                   0
  offloaded_doublevadd_publisher            XRT_FLAT  offloaded_doublevadd_publisher            XRT_FLAT               (0+0)                  -1

````

I can see the streamline and integrated dtbo, bin and xclbin files in the ros2_ws, but  ``` ros2 acceleration list ```  is not listing them even after restarting the dfx_mgr

vmayoral, 2022-07-17:
This is documented in prior examples, see https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/3_offloading_ros2_publisher.html. It might indeed be worth adding this if to perception tutorial if you bumped into it:

```bash
# restart the daemon that manages the acceleration kernels
$ ros2 acceleration stop; ros2 acceleration start
```

mohammedrafi-sk, 2022-07-18:
Hi @vmayoral 

I restarted the daemon, But only image_proc appears in the ```ros2 acceleration list```.  Even after the board reboot, only image_proc appears in the list.

The only way I could get the image_proc_streamlined and image_proc_integrated is with the workaround mentioned above

vmayoral, 2022-07-18:
Restarting the daemon manually is not enough, you need to copy the
firmware. That’s what those commands I mentioned above do.

jasvinderkhurana, 2022-09-13:
@mohammedrafi-sk, I hope this issue is resolved by the latest steps shall I close the issue?