Xilinx/KRS #67: KRS 1.0 review with KR260

vmayoral, 2022-07-17:
This ticket will discuss and review KRS `1.0` release using particulaly Ubuntu 22.04 and the cross-compilation development path ([instructions](https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#cross-compilation-development)). The goal of this review is to ensure instructions are reproducible.

Tickets related to this review that I'm tracking wherein users are reporting problems:
- https://github.com/Xilinx/KRS/issues/66
- https://github.com/Xilinx/KRS/issues/65
- https://github.com/Xilinx/KRS/issues/63
- https://github.com/Xilinx/KRS/issues/62
- https://github.com/Xilinx/KRS/issues/59
- https://github.com/ros-acceleration/acceleration_examples/issues/18

vmayoral, 2022-07-17:
Found an issue at which follows after https://github.com/Xilinx/KRS/pull/60:

```bash
###################################################
# 1. install some dependencies you might be missing
#
# NOTE: gcc-multilib conflicts with Yocto/PetaLinux 2022.1 dependencies
# so you can't have both paths simultaneously enabled in a single
# development machine
###################################################
sudo apt-get -y install curl build-essential libssl-dev git wget \
                          ocl-icd-* opencl-headers python3-vcstool \
                          python3-colcon-common-extensions python3-colcon-mixin \
                          kpartx u-boot-tools pv gcc-multilib gcc-aarch64-linux-gnu g++-aarch64-linux-gnu

```
This leads to:

```bash
The following packages have unmet dependencies:
 g++-aarch64-linux-gnu : Depends: g++-11-aarch64-linux-gnu (>= 11.2.0-1~)
 gcc-aarch64-linux-gnu : Depends: gcc-11-aarch64-linux-gnu (>= 11.2.0-1~)
E: Unable to correct problems, you have held broken packages.
```
which is related to https://github.com/Xilinx/KRS/pull/60/files, contributed by @mohammedrafi-sk. I think this needs to be double checked @jasvinderkhurana and @mohammedrafi-sk. On a second, closer review, to me, the following does it:

```bash
sudo apt-get -y install curl build-essential libssl-dev git wget \
                          ocl-icd-* opencl-headers python3-vcstool \
                          python3-colcon-common-extensions python3-colcon-mixin \
                          kpartx u-boot-tools pv gcc-multilib
```

(no need for https://github.com/Xilinx/KRS/pull/60 actually)

vmayoral, 2022-07-17:
Altogether, no further issues experienced  if taking into account https://github.com/Xilinx/KRS/issues/62. Since this seems to have caused some issues, I went ahead and submitted a PR to the KRS documentation that *introduces* this workaround into the docs: https://github.com/Xilinx/KRS/pull/68

I triple-checked ✅  and then recorded the process in separated shots (*due to restrictions of asciinema*):

| parts (1-5) | parts (6-7) | building an accelerator |
|----------|---------|----------|
| [image] | [image] | [image] |

vmayoral, 2022-07-17:
All good in here. Tickets below should also be resolved by maintainers since this confirms functionality:

> - https://github.com/Xilinx/KRS/issues/66
> - https://github.com/Xilinx/KRS/issues/65
> - https://github.com/Xilinx/KRS/issues/63
> - https://github.com/Xilinx/KRS/issues/62
> - https://github.com/Xilinx/KRS/issues/59
> - https://github.com/ros-acceleration/acceleration_examples/issues/18
>

closed 2022-07-17 by vmayoral

wknitter, 2022-07-27:
Where did the KV260 firmware come from in your /tmp directory in your parts (6-7) video? I've tried following your steps exactly several times and I have to wget the kv260 firmware manually every time. 
So this never works for me:
~/krs_ws$ mv /tmp/acceleration_firmware_kv260 ./src/firmware
I have to do this:
~/krs_ws/src/firmware$ wget https://www.xilinx.com/bin/public/openDownload?filename=acceleration_firmware_kv260.zip

Also, I have to run a sudo command right before running the first colcon build command or it hangs indefinitely on step 32 of 33. How did you get the workspace to build without entering your sudo password (since it never actually asks for it like the install instructions say [here](https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#cross-compilation-development))? 
For reference, here's what I have to do to keep the build from going on indefinitely in step 32 of 33:
~/krs_ws$ sudo ls -la
[enter sudo password]
~/krs_ws$ colcon build --merge-install

vmayoral, 2022-07-28:
@wknitter I believe this is clarified at https://asciinema.org/a/509153?t=1:44. For reference, have a look at the PR https://github.com/Xilinx/KRS/pull/68 listed above. That should have everything necessary to reproduce it.

This is confirmed to work but It's up to Xilinx's team to evaluate it and merge this upstream.

> 
>  For reference, here's what I have to do to keep the build from going on indefinitely in step 32 of 33:
> ~/krs_ws$ sudo ls -la
> [enter sudo password]
> ~/krs_ws$ colcon build --merge-install

I think this is good input. @mohammedrafi-sk, @jasvinderkhurana and others in the Xilinx team, you may want to consider @wknitter's comment above.

jasvinderkhurana, 2022-07-29:
@vmayoral, I reviewed your Pull request and had one question which I posted in review comment and was waiting for your response on it. I am not sure if you are able to see my comments. See [here](https://github.com/Xilinx/KRS/pull/68/files/c663076e16ad7f51378373f0db01d6fdc8936b44).

Comment: 
```
I have one confusion here, I am assuming it is building for KR260 target, but still we are fetching kv260 firmware. May be it will be confusing, wouldn't it be better if this all be in kr260_firmware?
```

@wknitter, thanks for your input, I personally did not face issue without adding sudo command, but I will check again and will add it if required.

jasvinderkhurana, 2022-07-29:
@vmayoral, Sorry my mistake, I inserted the review comments but never submitted review. So you did not see it. But please have a look if my question is valid.

wknitter, 2022-07-29:
@vmayoral I reviewed that part of the video again, but there is no explanation how the KV260 firmware ended up in your /tmp directory and it does not appear in my /tmp directory at any point while following your steps. I've followed everything in PR #68 exactly. 

Also, the package gcc-multilib and gcc-aarch64-linux-gnu g++-aarch64-linux-gnu cannot be passed to the same apt install command because one replaces the other which is why passing them to the same apt install command results in broken packages. 

The gcc-multilib package is the wrong GNU C compiler for the cross compilation of Ubuntu and needs to be removed from that set of instructions for the KR260 altogether. The catch though is, you have to reinstall the gcc-multilib package when you switch back over to targeting the KV260 to build a PetaLinux root filesystem for it (which uninstalls the gcc-aarch64-linux-gnu g++-aarch64-linux-gnu packages automatically, so you have to reinstall them again to switch back to the KR260).

vmayoral, 2022-07-29:
> @vmayoral I reviewed that part of the video again, but there is no explanation how the KV260 firmware ended up in your /tmp directory and it does not appear in my /tmp directory at any point while following your steps. I've followed everything in PR https://github.com/Xilinx/KRS/pull/68 exactly.

Once again, read https://github.com/Xilinx/KRS/issues/67#issuecomment-1186434852 and see the video. There's a comment in the video hinting that you should be fetching it the way you were. Nothing wrong in there. Same's explained in https://github.com/Xilinx/KRS/pull/68. See particularly https://github.com/Xilinx/KRS/pull/68/files#diff-e7486e0e889e10f1846dbe8d85f945138be6e23dff3aee984ea39751b476cf93R303 and https://github.com/Xilinx/KRS/pull/68/files#diff-e7486e0e889e10f1846dbe8d85f945138be6e23dff3aee984ea39751b476cf93R304.

> Also, the package gcc-multilib and gcc-aarch64-linux-gnu g++-aarch64-linux-gnu cannot be passed to the same apt install command because one replaces the other which is why passing them to the same [...]

I can't confirm this. Instructions as provided in #68 work for me.

wknitter, 2022-07-29:
@vmayoral Ah I misunderstood that originally, my apologies. I thought it implied that the firmware should have already been in the /tmp directory.

Take a look at the terminal output at [0:24](https://asciinema.org/a/509152) of your parts 1-5 video, the apt command says it has to remove gcc-multilib to install gcc-aarch64-linux-gnu and g++-aarch64-linux-gnu.

jasvinderkhurana, 2022-08-02:
@vmayoral, I had merged your pull request #68 , but I am seeing the sphinx build error in actions. I have already created an issue #69, can you please have a look?