Xilinx/KRS #69: Auto build failed after new pull request merge

jasvinderkhurana, 2022-07-30:
@vmayoral,

I have merged your [pull request #68](https://github.com/Xilinx/KRS/pull/68), and I see [Build and Deploy](https://github.com/Xilinx/KRS/runs/7589204947?check_suite_focus=true) error in Actions, can you please have a look? 
I get below error:

```
Run cd sphinx; make html
Running Sphinx v4.0.2
loading pickled environment... failed
failed: source directory has changed
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 19 source files that are out of date
updating environment: [new config] 19 added, 0 changed, 0 removed
reading sources... [  5%] docs/examples/0_ros2_publisher

Extension error (sphinx_markdown_tables):
Handler <function process_tables at 0x7f073a47c1[6](https://github.com/Xilinx/KRS/runs/7589204947?check_suite_focus=true#step:5:7)0> for event 'source-read' threw an exception (exception: __init__() missing 1 required positional argument: 'config')
make: *** [Makefile:20: html] Error 2
Error: Process completed with exit code 2.
```

vmayoral, 2022-09-07:
I've been some time trying to reproduce this locally, in my usual dev machine I can't reproduce such issue:

<details><summary>✅ OS X 12.3.1 (21.4.0 Darwin Kernel Version 21.4.0: Fri Mar 18 00:46:32 PDT 2022; root:xnu-8020.101.4~15/RELEASE_ARM64_T6000 arm64)</summary>

```bash
victor@accelerationrobotics:/tmp/KRS/sphinx$ make html
Ejecutando Sphinx v4.0.2
cargando el ambiente pickled... hecho
compilando [mo]:los objetivos para 0 los archivos po que estan desactualizados
compilando [html]:los objetivos para 0 los archivos fuentes que estan desactualizados
actualizando ambiente[configuración modificada ('recommonmark_config')] 19añadido, 0 cambiado, 0 removido
/usr/local/lib/python3.9/site-packages/recommonmark/parser.py:75: UserWarning: Container node skipped: type=document
  warn("Container node skipped: type={0}".format(mdnode.t))
leyendo fuentes...[100%] index
/private/tmp/KRS/sphinx/source/docs/examples/6_perception.md:1: WARNING: Content block expected for the "sidebar" directive; none found.
/private/tmp/KRS/sphinx/source/docs/examples/graphs.rst:1: WARNING: Title overline too short.

####################
  Accelerating ROS 2 Graphs
####################
/private/tmp/KRS/sphinx/source/docs/examples/nodes.rst:1: WARNING: Title overline too short.

####################
  Accelerating ROS 2 Nodes
####################
/private/tmp/KRS/sphinx/source/docs/howto.md:2: WARNING: Pie de página [1] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:3: WARNING: Pie de página [2] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:4: WARNING: Pie de página [3] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:5: WARNING: Pie de página [4] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:6: WARNING: Pie de página [5] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:7: WARNING: Pie de página [6] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:8: WARNING: Pie de página [7] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:9: WARNING: Pie de página [8] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:10: WARNING: Pie de página [9] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:11: WARNING: Pie de página [10] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:12: WARNING: Pie de página [11] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:13: WARNING: Pie de página [12] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:14: WARNING: Pie de página [13] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:15: WARNING: Pie de página [14] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:16: WARNING: Pie de página [15] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:17: WARNING: Pie de página [16] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:18: WARNING: Pie de página [17] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:19: WARNING: Pie de página [18] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:20: WARNING: Pie de página [19] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:21: WARNING: Pie de página [20] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:22: WARNING: Pie de página [21] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:23: WARNING: Pie de página [22] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:24: WARNING: Pie de página [23] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:25: WARNING: Pie de página [24] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:26: WARNING: Pie de página [25] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:27: WARNING: Pie de página [26] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:28: WARNING: Pie de página [27] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:29: WARNING: Pie de página [28] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:30: WARNING: Pie de página [29] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:31: WARNING: Pie de página [30] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:32: WARNING: Pie de página [31] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:33: WARNING: Pie de página [32] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/howto.md:34: WARNING: Pie de página [33] no está referenciado.
/private/tmp/KRS/sphinx/source/docs/install.md:1: WARNING: Duplicate explicit target name: "install instructions".
/private/tmp/KRS/sphinx/source/docs/install.md:1: WARNING: Duplicate explicit target name: "install instructions".
buscando por archivos no actualizados...no encontrado
preparando ambiente... hecho
verificando consistencia... hecho
preparando documentos... hecho
escribiendo salida... [100%] index
/private/tmp/KRS/sphinx/source/docs/examples/1_hello_xilinx.md:172: WARNING: No pudo el léxico literal_block como "bash". Destacado omitido.
/private/tmp/KRS/sphinx/source/docs/howto.md:1237: WARNING: None:any reference target not found: #
/private/tmp/KRS/sphinx/source/docs/howto.md:14: WARNING: No pudo el léxico literal_block como "bash". Destacado omitido.
/private/tmp/KRS/sphinx/source/docs/howto.md:443: WARNING: No pudo el léxico literal_block como "bash". Destacado omitido.
[... cut ...]
RUN apt-get update && export DEBIAN_FRONTEND=noninteractive \
    && apt-get -y install --no-install-recommends \
        python3 python3-pip git build-essential

# SUCEEDS
RUN pip3 install rst2pdf==0.99 sphinx==5.1.1 recommonmark==0.7.1 sphinx-markdown-tables==0.0.17 docutils==0.19

# FAILS
# RUN pip3 install rst2pdf==0.98 recommonmark==0.7.1 sphinx_markdown_tables==0.0.15 sphinx==4.0.2 docutils==0.16

RUN git clone https://github.com/Xilinx/KRS
RUN cd KRS/sphinx; make html
```

</details>

This last iteration builds just fine:

[image]

jasvinderkhurana, 2022-09-08:
@vmayoral Thanks! closing the issue as the auto build is successful

closed 2022-09-08 by jasvinderkhurana