Xilinx/KRS #73: Issue with SD image generated following ROS 2 publisher example

mvgianel, 2022-10-13:
I followed example 0 to create an SD and use it on the KV260 but instead of getting 
xilinx-k26-starterkit-2020_2 login:

I got 
/ # 

After reseting the board while keeping it connected to my computer I got the following messages:
[image]

I am used Balena Etcher and the linux method described here: https://www.xilinx.com/products/som/kria/kv260-vision-starter-kit/kv260-getting-started/setting-up-the-sd-card-image.html but got the same result.

Does anyone know what could be causing this?

vmayoral, 2022-10-13:
Btw, the issue above is that the kernel's not finding the rootfs while running `init`. This is typically because your boot script (which lives in the first partition of your SD card) is not properly configured and/or your device tree blob doesn't match appropriately.

You could probably boot successfully fixing this. If you're not familiar with these concepts though, I'd recommend checking out the *Ubuntu* path of KRS's release. See https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#ubuntu-22-04.

mvgianel, 2022-10-14:
> Another possibility is that you're not using the right resources. What confuses me a bit though is that you above report:
> 
> > xilinx-k26-starterkit-2020_2

That was what the example in the Documentation said I should get. What I actually get is "# /" as shown in the picture. I am using Vitis 2022.1 and Ubuntu 22.04 to develop my applications and trying to follow the flow of creating the SD image with colcon acceleration.

> The concepts WILL remain the same and the dev flow identical to the usual ROS one, but moving to Ubuntu will provide a more developer friendly environment. Also, instead of having to create your own SD card image, you'll only need to flash Ubuntu's official Kria images (see [here](https://ubuntu.com/download/amd-xilinx)).

Does that mean that only the Native development will be supported in the future?

> You could probably boot successfully fixing this. If you're not familiar with these concepts though, I'd recommend checking out the _Ubuntu_ path of KRS's release. See https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#ubuntu-22-04.

I followed these steps in my development platform and then tried to run this example https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/0_ros2_publisher.html#building-creating-the-raw-image-and-running-in-hardware, but the image created using that approach results in my issue

jasvinderkhurana, 2022-10-14:
When you are upgrading to Vitis 2022.1 with Ubuntu 22.04, you might need to upgrade the firmware as well, Please go through the documentation below. On how to update firmware.

https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/1641152513/Kria+K26+SOM#Boot-Firmware-Updates

vmayoral, 2022-10-14:
> That was what the example in the Documentation said I should get. What I actually get is "# /" as shown in the picture. I am using Vitis 2022.1 and Ubuntu 22.04 to develop my applications and trying to follow the flow of creating the SD image with colcon acceleration.

@mvgianel then follow guidelines for Ubuntu, not for Yocto. See https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#ubuntu-22-04.

> Does that mean that only the Native development will be supported in the future?

No, with Ubuntu you've got three avenues of development:
- [Native (on-target/board) development ](https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#native-on-target-development)
- [Cross-compilation development](https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#cross-compilation-development)
- [QEMU (emulation) development](https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#qemu-emulation-development)

mvgianel, 2022-10-14:
> When you are upgrading to Vitis 2022.1 with Ubuntu 22.04, you might need to upgrade the firmware as well, Please go through the documentation below. On how to update firmware.
> 
> https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/1641152513/Kria+K26+SOM#Boot-Firmware-Updates

I had upgraded the firmware but I just noticed there is a newer version. I will upgrade and report later if that solves the issue.

mvgianel, 2022-10-14:
@jasvinderkhurana after upgrading the firmware, I was able to use the official Kria Ubuntu 22.04, but I still have the same issue with the image created by following the tutorial.

ohkawatks, 2022-11-05:
I have succeeded in booting KRS1.0 (petalinux) on KR260 from sdcard.
This is a temporary workaround, however, I hope this helps someone.

Development Environment: 
Ubuntu22.04, Vitis2022.1, Petalinux2022.1

***

Firstly I have finished the sdcard image creation by the following two instructions:
https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#ubuntu-22-04
and
https://xilinx.github.io/KRS/sphinx/build/html/docs/examples/0_ros2_publisher.html

(I have ignored the error while the whole build "$ colcon build --merge-install". And I have just built the KRS1.0 sdcard image partly by "$ colcon build --build-base=build-kv260 --install-base=install-kv260 --merge-install --mixin kv260 --packages-select ament_acceleration ament_vitis vadd_publisher")

Then, I have done the two steps after burning the sdcard image:

1. Overwrite the **system.dtb** in the boot partition of the sdcard.
2. Add **uEnv.txt** to overwrite bootargs for the linux boot (rootfs=/dev/sda2, and so on)

***

Currently (KRS1.0) the **system.dtb** is for KV260. So I have copied the **system-zynqmp-sck-kr-g-revB.dtb** file as **system.dtb** into the boot partition. After replacing the **system.dtb**, device **/dev/sda2** appears in the linux kernel message.

The file was obtained from the petalinux2022.1 + KR260 BSP:
```
$ petalinux-create --type project -s xilinx-kr260-starterkit-v2022.1-05140151.bsp --name kr260_linux
```
Then I could find **system-zynqmp-sck-kr-g-revB.dtb** in the pre-built/linux/images directory.

***

I have used uEnv.txt written below:

```
linux_img_load_cmd=fatload usb 0:1 0x00200000 Image
linux_fdt_load_cmd=fatload usb 0:1 0x00100000 system.dtb
linux_ramdisk_load_cmd=fatload usb 0:1 0x04000000 ramdisk.cpio.gz.u-boot

linux_boot_cmd=booti 0x00200000 - 0x00100000
linux_load_cmd=env run linux_img_load_cmd && env run linux_fdt_load_cmd
linux_args_cmd=setenv bootargs earlycon root=/dev/sda2 rootwait console=ttyPS1,115200 clk_ignore_unused xilinx_tsn_ep.st_pcp=4 init_fatal_sh=1 cma=900M

uenvcmd=env run linux_args_cmd && env run linux_load_cmd && env run linux_boot_cmd
```

The error suffering us was "/dev/mmcblk1p2 is not found". After replacing the system.dtb, /dev/sda2 can be accessed. Then the uEnv.txt file skips the two-staged boot and directly boot using root=/dev/sda2.

***

Integrating this fix into firmware/acceleration_firmware_kr260 is another issue.

mvgianel, 2022-11-18:
@ohkawatks Thank you so much for helping! I am working with a KV260 and Ubuntu 22.04 but I followed the steps you delined using the petalinux kv260 image. I also had to use system-zynqmp-sck-kv-g-revB.dbt but y renamed it to system.dbt. However, after following the steps I get the same behavior as I previously mentioned

ohkawatks, 2022-11-26:
Can you see /dev/sda, /dev/sda1, and /dev/sda2 after the error?
If you cannot see them, it should be a problem of system.dtb.
I noticed that the filename you wrote is system.dbt. Please name it system.dtb. (Device Tree Blob)

Applepi, 2022-12-06:
This is still broken https://support.xilinx.com/s/question/0D54U00005zHNQHSA4/sd-card-image-from-krs-github-tutorial-does-not-boot?language=en_US&t=1670337200519

lanyangyang, 2022-12-08:
I have encountered the same issue in KR260.

lanyangyang, 2022-12-13:
I found the simplest solution so far, but on kr260. 
1. Use the preloaded ubuntu22.04 system, after booting, manually install ros2-humble-desktop-full.
2. Then copy the colcon-built folder on the host (for example, install-kr260) to kr260 (for example, home/ubuntu/install-kr260). 3. Connect kr260 through putty, then run the following command,
`source /opt/ros/humble/setup.bash`
`source install-kr260/local_setup.bash`
After this step, u can see that vadd_publisher is in the ros2 pkg list. 
Hope this works for you too.

mvgianel, 2023-01-26:
@lanyangyang 

Thank you for the tip! I tried this but I run into this issue when I try to run a node: 
[image]
Have you gotten past

mvgianel, 2023-01-26:
Hi @lanyangyang,

Thank you for your reply! I was able to get everything working after following your instructions :)

mvgianel, 2023-01-26:
Hi @ohkawatks,

Sorry, I misspelled the file extension here, but it was correct on the board. I was not able to see /dev/sda after adding the uEnv.txt and changing the system.dtb on the sd card. However, after following the instructions bellow I was able to log into petalinux:

[image]

However, I cannot locate the ros files that I need to source to run my code:

paholden, 2023-04-10:
Great to know for PetaLinux.  Is there any way to set up FreeRTOS to boot from SD card?

sachinkum0009, 2024-01-01:
@mvgianel 

you need to modify the partition name to mmcblk0p2 in the uEnv.txt  in order to locate the ros files.

> Hi @ohkawatks,
> 
> Sorry, I misspelled the file extension here, but it was correct on the board. I was not able to see /dev/sda after adding the uEnv.txt and changing the system.dtb on the sd card. However, after following the instructions bellow I was able to log into petalinux:
> 
> [image]
> 
> However, I cannot locate the ros files that I need to source to run my ros2 node:
> 
> [image]
> 
> Do you know what could be causing this?