Xilinx/KRS #74: Failure in compiling doublevadd_publisher

jasvinderkhurana, 2022-10-14:
Hi,

I have tried to setup a fresh on new ubuntu machine and install KRS, but it is failing to compile at accelerated_doublevadd_publisher. I have used below steps:

```
Install new ubuntu image 22.04
Install Vitis 2022.1 tools

#Install ROS2 and Gazebo
sudo apt install ros-humble-desktop-full
sudo apt-get install ros-humble-gazebo-ros-pkgs

#Install KRS
sudo apt-get -y install curl build-essential libssl-dev git wget \
                          ocl-icd-* opencl-headers python3-vcstool \
                          python3-colcon-common-extensions python3-colcon-mixin \
                          kpartx u-boot-tools pv gcc-multilib
sudo apt-get -y install gcc-aarch64-linux-gnu g++-aarch64-linux-gnu
sudo apt-get install qemu-user-static
mkdir -p ~/krs_ws/src; cd ~/krs_ws

cat << 'EOF' > krs_humble.repos
repositories:  
  perception/image_pipeline:
    type: git
    url: https://github.com/ros-acceleration/image_pipeline
    version: ros2

  tracing/tracetools_acceleration:
    type: git
    url: https://github.com/ros-acceleration/tracetools_acceleration
    version: humble

  firmware/acceleration_firmware_kr260:
    type: zip
    url: https://github.com/ros-acceleration/acceleration_firmware_kr260/releases/download/v1.0.0/acceleration_firmware_kr260.zip

  acceleration/adaptive_component:
    type: git
    url: https://github.com/ros-acceleration/adaptive_component
    version: humble
  acceleration/ament_acceleration:
    type: git
    url: https://github.com/ros-acceleration/ament_acceleration
    version: humble
  acceleration/ament_vitis:
    type: git
    url: https://github.com/ros-acceleration/ament_vitis
    version: humble
  acceleration/colcon-hardware-acceleration:
    type: git
    url: https://github.com/colcon/colcon-hardware-acceleration
    version: main
  acceleration/ros2_kria:
    type: git
    url: https://github.com/ros-acceleration/ros2_kria
    version: main
  acceleration/ros2acceleration:
    type: git
    url: https://github.com/ros-acceleration/ros2acceleration
    version: humble
  acceleration/vitis_common:
    type: git
    url: https://github.com/ros-acceleration/vitis_common
    version: humble
  acceleration/acceleration_examples:
    type: git
    url: https://github.com/ros-acceleration/acceleration_examples
    version: main
EOF

vcs import src --recursive < krs_humble.repos
source /tools/Xilinx/Vitis/2022.1/settings64.sh  # source Xilinx tools
source /opt/ros/humble/setup.bash  # Sources system ROS 2 installation.

export PATH="/usr/bin":$PATH 
colcon build --merge-install

```

Below is the error occured:

[image]

@vmayoral, can you please have a quick look at it.. 

PS: The KRS used to build earlier may be 1 month back, but is failing now.

jasvinderkhurana, 2022-10-14:
The root cause seems to be ROS_VITIS macro is somehow not getting enabled and then it is not installing the vitis_common headers in install directory. using below command it succeeds the doublevadd_publisher, but fails in another package:

```
colcon build  --cmake-args -D ROS_VITIS=1  --merge-install 

```
Here is the error:

[image]

But then I am currently skipping that package to proceed the build

```
colcon build  --cmake-args -D ROS_VITIS=1  --merge-install  --packages-select ros2_kria_power

```

vmayoral, 2022-10-14:
@jasvinderkhurana I'm traveling and can't have a look at your issues right now in detail. My quick feedback is that KRS is known to build. See CI pipeline https://gitlab.com/accelerationrobotics/customers/amd/krs-pipeline/-/pipelines/660158697. Inspect the corresponding files at that point if you need further help.

My suggestion is for you to hold until the CI stabilizes (still in development) and build on top of that (which is reproducible) once ready to avoid uncertainty. It'll take some weeks still.

verderog, 2022-10-25:
I'm trying to build using Docker via the Ubuntu 22.04 instructions listed [here](https://github.com/Xilinx/KRS/blob/main/sphinx/source/docs/install.md) and am running into this same issue.  Here's the error that I see:

```
--- stderr: accelerated_vadd
/krs_ws/src/acceleration/acceleration_examples/nodes/accelerated_vadd/src/host.cpp:24:10: fatal error: vitis_common/common/ros_opencl_120.hpp: No such file or directory
   24 | #include <vitis_common/common/ros_opencl_120.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/accelerated_vadd.dir/build.make:76: CMakeFiles/accelerated_vadd.dir/src/host.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/accelerated_vadd.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< accelerated_vadd [5.04s, exited with code 2]

```

I just got my KR260 dev kit last week and this is the first time I've tried to build the KRS.  

Further, I'm getting a 404 when I try to follow the CI pipeline link @vmayoral has referenced in https://github.com/Xilinx/KRS/issues/74#issuecomment-1278999392.  So, I can't compare how that's working versus how my build is not.

jasvinderkhurana, 2022-10-25:
@verderog , thanks for confirming, we are working towards making KRS more stable. So please stay tuned

verderog, 2022-10-27:
I've been able to progress through much of the [Ubuntu 22.04 Install Instructions](https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html).  However, when I hit step 7.A, I encounter several `colcon` build errors.  Here's the first one:

```
CMake Error at CMakeLists.txt:12 (find_package):
  By not providing "Findament_cmake_core.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_core", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake_core"
  with any of the following names:

    ament_cmake_coreConfig.cmake
    ament_cmake_core-config.cmake

  Add the installation prefix of "ament_cmake_core" to CMAKE_PREFIX_PATH or
  set "ament_cmake_core_DIR" to a directory containing one of the above
  files.  If "ament_cmake_core" provides a separate development package or
  SDK, be sure it has been installed.

```

The others are similar, but reference different packages such as `ament_cmake`, `ament_cmake_ros` and `ament_cmake_auto`.  

I'm executing the instructions from the Xilinx page linked above in order.  I've googled potential causes for the `ament_cmake_core` problem and they mention that my ROS2 Humble `setup.bash` could be the culprit, but I've made sure to `source` that as well as the `install/setup.bash` from the build.

I do see `/opt/ros/humble/share/ament_cmake_core/cmake/ament_cmake_coreConfig.cmake`, so I don't know why cmake is not finding it correctly.  I've tried specifying `-DCMAKE_PREFIX_PATH=/opt/ros/humble/share` but still see the same failure.

If I explicitly define the location of ament_cmake_core via `-Dament_cmake_core_DIR="/opt/ros/humble/share/ament_cmake_core/cmake"`, then I can get `ament_accleration` to build.  This seems like overkill and something else is fundamentally wrong with my environment or the build configuration.

Any ideas or suggestions on what may be causing the problem?

satoru-i, 2022-10-28:
I edited krs_humble.repos and changed vitis_common version from humble to master. After that I was able to build successfully. Is this related to this error?

jasvinderkhurana, 2022-11-05:
@vmayoral, I am also facing similar issue, and after updating the vitis_common package to master branch. That also did not resolved my problem.

Can you please check, if there is any latest change that broke the build?

lanyangyang, 2022-12-05:
@jasvinderkhurana  I think you can try to delete the previous build folder and build again. I also encountered the same problem, and after about three rebuilds, the problem was solved.