Xilinx/KRS #78: Missing file when running doublevadd_publisher on Ubuntu 22.04

lanyangyang, 2022-12-16:
I scp my workspace to KR260. Then, I sourced the correct file and ran krs_ws/local_setup.bash. So far no issues and no error messages. Then, when I used `ros2 run doublevadd_publisher doublevadd_publisher`, I got the following message,
[image]

In the previous built period, I selected kv260 firmware and used --mixin kr260. It is weird that I can run vadd_publisher at 10Hz successfully.
Can anyone give me some advice?

jasvinderkhurana, 2022-12-16:
@lanyangyang, Thanks for trying this. I am not sure what is causing this, but I will try to debug at myside, also in the meantime. can you please share the complete steps to reproduce your issue?

lanyangyang, 2022-12-17:
Thanks for your reply. @jasvinderkhurana 

My steps are as follows,
First of all, I followed [https://china.xilinx.com/products/som/kria/kr260-robotics-starter-kit/kr260-getting-started/getting-started.html] this tutorial to install Ubuntu 22.04 on KR260. Then, I installed ros2-humble-desktop-full. Also, I have upgraded the firmware.

Then, I followed [https://github.com/Xilinx/KRS/blob/main/sphinx/source/docs/install.md] this tutorial to install KRS on my host computer. During this step, I created two workspaces, one with kv260 firmware and one with kr260 firmware, but both used `--mixin kr260` in colcon build.

After these steps, I tried ROS2 examples as [https://github.com/Xilinx/KRS/blob/main/sphinx/source/docs/examples/0_ros2_publisher.md]
In workspace one, I used `colcon acceleration select kv260` and `--mixin kr260`. Then, `scp` the workspace to kr260.  Then, when I used `ros2 run doublevadd_publisher doublevadd_publisher`, I got the above message.

Thus, I decided to try with workspace two as I thought this issue might was caused by firmware. I used `colcon acceleration select kr260` and `--mixin kr260`. Then, `scp` the workspace to kr260.  Then, when I used `ros2 run doublevadd_publisher doublevadd_publisher`, I got the exactly the same message.