Xilinx/KRS #80: KRS not compiling

macshouldbeusedforgaming, 2023-01-16:
When I tried to build everything together using colcon build --merge-install, I came up with this error. I am using Vitis 2022.2 and ROS 2 Humble.
Starting >>> ament_acceleration
Starting >>> tracetools_image_pipeline
Starting >>> tracetools_acceleration
Starting >>> adaptive_component
Finished <<< ament_acceleration [0.49s]                                   
Starting >>> ament_vitis
Finished <<< tracetools_image_pipeline [0.63s]
Starting >>> camera_calibration
Finished <<< tracetools_acceleration [0.66s]
Starting >>> depth_image_proc
Finished <<< adaptive_component [0.70s]                                   
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' instead
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
  warnings.warn(
Starting >>> image_publisher                                       
Finished <<< ament_vitis [0.65s]                                   
Starting >>> vitis_common
--- stderr: image_publisher                                       
/home/urmother/krs_ws/src/perception/image_pipeline/image_publisher/src/image_publisher.cpp:34:10: fatal error: camera_info_manager/camera_info_manager.hpp: No such file or directory
   34 | #include <camera_info_manager/camera_info_manager.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/image_publisher.dir/build.make:76: CMakeFiles/image_publisher.dir/src/image_publisher.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/image_publisher.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< image_publisher [0.38s, exited with code 2]
Aborted  <<< vitis_common [0.30s]                                            
Aborted  <<< depth_image_proc [0.77s]
Aborted  <<< camera_calibration [1.02s]                      

Summary: 5 packages finished [2.20s]
  1 package failed: image_publisher
  3 packages aborted: camera_calibration depth_image_proc vitis_common
  1 package had stderr output: image_publisher
  24 packages not processed

jasvinderkhurana, 2023-02-03:
Hi KRS is tested with Vitis 2022.1 tools, can you please give a try with 2022.1 version?

jasvinderkhurana, 2023-03-29:
@macshouldbeusedforgaming, can you please share if your compilation is successful with 2022.1 tools?

jasvinderkhurana, 2023-07-27:
Closing the ticket for no response for long...

closed 2023-07-27 by jasvinderkhurana