Xilinx/KRS #84: KR260 is not officially supported? Any alternative for Hardware Accelaration with Yocto ROS2?

nikAizuddin, 2023-05-14:
I don't see KR260 in the Supported Hardware list. Is there any instructions how to use Hardware Acceleration with Yocto ROS2?

vmayoral, 2023-05-14:
> I don't see KR260 in the Supported Hardware list.

Which "list" are you referring to? 

On enabling KR260 with Yocto and instructions on it, check out [these guidelines](https://news.accelerationrobotics.com/ros-2-humble-in-amd-kr260/) to get ROS 2 Humble in AMD KR260 with Yocto however note there's limited support. Unless you have enough experience with Yocto, I'd recommend instead looking into the officially documented pathway of [using Ubuntu](https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html#ubuntu-22-04).

nikAizuddin, 2023-05-14:
> Which "list" are you referring to?

I'm referring to this list https://github.com/Xilinx/KRS/blob/main/sphinx/source/docs/hardware.md.

I have successfully build ROS2 Humble with PetaLinux 2022.2 on my KR260. (I have noted that KRS only support 2022.1).

According to your guidelines, [ROBOTCORE](https://accelerationrobotics.com/robotcore.php) is the only way to use KR260 accelaration with ROS2 PetaLinux. Am I correct?

vmayoral, 2023-05-14:
> I'm referring to this list https://github.com/Xilinx/KRS/blob/main/sphinx/source/docs/hardware.md.

Right, thanks, that list should be updated. Ping @jasvinderkhurana.

> According to your guidelines, [ROBOTCORE](https://accelerationrobotics.com/robotcore.php) is the only way to use KR260 accelaration with ROS2 PetaLinux. Am I correct?

No. You can do it yourself easily as well. [ROBOTCORE Framework](https://accelerationrobotics.com/robotcore-framework.php) presents an alternative with additional hardware accelerated capabilities. If you're looking just for vanilla ROS 2 support, you should use AMD/Xilinx's PetaLinux and meta-ros ([this branch](https://github.com/ros/meta-ros/tree/honister) for honister).

nikAizuddin, 2023-05-15:
> No. You can do it yourself easily as well. [ROBOTCORE Framework](https://accelerationrobotics.com/robotcore-framework.php) presents an alternative with additional hardware accelerated capabilities. [...]

I see, thank you for clarifying. I've been using meta-ros Humble provided by PetaLinux 2022.2. I have searched `components/yocto/layers/meta-ros/meta-ros2-humble/generated-recipes/` and found most repositories from [ros-acceleration](https://github.com/ros-acceleration/) are already included.

closed 2023-05-15 by nikAizuddin

jasvinderkhurana, 2023-05-15:
@vmayoral @nikAizuddin , Thanks for pointing out of date page of Hardware support. I will get this updated.