Xilinx/KRS #91: Failed to cross compile for kr260

samcts2309, 2023-08-01:
I failed to build all ros2 packages which requires `/usr/lib/aarch64-linux-gnu/libpython3.10.so` during the cross compilation.

Code Execute:
```
###################################################
# 7.A cross-compile and generate ONLY CPU binaries
###################################################
colcon build --build-base=build-kr260-ubuntu --install-base=install-kr260-ubuntu --merge-install --mixin kr260 --cmake-args -DNOKERNELS=true
```

Error Msg:
```
--- stderr: adaptive_component
gmake[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libpython3.10.so', needed by 'libadaptive_component.so'.  Stop.
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/adaptive_component.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< adaptive_component [0.47s, exited with code 2]
```

jasvinderkhurana, 2023-08-02:
@samcts2309,

Thanks a lot for trying this and pointing out. This seems to be a known issue, with Ubuntu rootfs as:

Ubuntu 22.04 sysroot is not fully prepared for cross-compilation (but for native builds instead) and thereby, while invoking FindPython, it's just picking the host resources, instead of the target/sysroot ones, which leads to ROS 2 packages relying on Python 3 getting a dependency against the host (which doesn't exist), instead of against the sysroot.
    A **workaround** for this is symlinking the Python3 library of the host to the sysroot one, so that it gets picked while cross-compiling against the Ubuntu 22.04 sysroot. The following should do: `sudo ln -s ~/krs_ws/install/../acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/libpython3.10.so.1.0 /usr/lib/aarch64-linux-gnu/libpython3.10.so`

Unknowingly this workaround got deleted from the documentation in last update. Thanks again for pointing out, I will update the documentation with above workaround. Also please let me know if this resolves your problem

jasvinderkhurana, 2023-08-02:
Closing it with pull request #92

closed 2023-08-02 by jasvinderkhurana