Xilinx/KRS #94: Could not find a package configuration file provided by "Qt5Widgets".

satoru-i, 2023-08-10:
I tried cross-compiling turtlesim for KR260 to get used to using the KRS.
https://github.com/ros/ros_tutorials/tree/humble-devel/turtlesim

However, when I try to build with the following command, I get an error.

```
colcon build --build-base=build-kr260-ubuntu --install-base=install-kr260-ubuntu --merge-install --mixin kr260 --packages-select turtlesim
``` 
``` 
--- stderr: turtlesim                         
CMake Error at /home/satorui/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/Qt5/Qt5Config.cmake:28 (find_package):
  Could not find a package configuration file provided by "Qt5Widgets" with
  any of the following names:

    Qt5WidgetsConfig.cmake
    qt5widgets-config.cmake

  Add the installation prefix of "Qt5Widgets" to CMAKE_PREFIX_PATH or set
  "Qt5Widgets_DIR" to a directory containing one of the above files.  If
  "Qt5Widgets" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:15 (find_package)

---
Failed   <<< turtlesim [0.91s, exited with code 1]
``` 
Qt5Widgets exists in sysroot, but how can I build an app that uses Qt5Widgets?

vmayoral, 2023-08-10:
Are you sure `Qt5Widgets` is properly set up in the sysroot? How have you verified it? If properly set in the sysroot, then, in principle, you wouldn't have this issue.

jasvinderkhurana, 2023-08-11:
@vmayoral, It seems QT is not set by default in rootfs, so can you confirm that in the KR260 firmware, did you enabled QT explicitly ? If not, then there might not be any QT libs. Can you guide on how to create a new firmware with QT enabled in sysroot?

vmayoral, 2023-08-12:
> @vmayoral, It seems QT is not set by default in rootfs, so can you confirm that in the KR260 firmware, did you enabled QT explicitly ? If not, then there might not be any QT libs. Can you guide on how to create a new firmware with QT enabled in sysroot?

QT's not part-of (or generally needed) and an embedded ROS rootfs. It implies a significant amount of deps. and adds unnecessary weight, unless you need it, as it seems the case above. This can of course be added by re-creating the firmware artifacts again.

Happy to work together with you and the team @jasvinderkhurana to recreate the firmware artifacts and include what's needed. Let's take this offline to a separate channel. Very excited  and discuss this in a next collaboration.

satoru-i, 2023-08-21:
Hi @vmayoral, @jasvinderkhurana,
Thank you both for your comments. As you've probably noticed, I installed KRS exactly as instructed, so I'm using the sysroot found in the firmware below.

https://github.com/ros-acceleration/acceleration_firmware_kr260/releases/download/v1.1.1/acceleration_firmware_kr260.zip

I would appreciate it if you could point me to "how to create a new firmware with QT enabled in sysroot".

jasvinderkhurana, 2023-08-21:
@satoru-i , Can you please confirm whether you are using Ubuntu flow or Petalinux/Yocto flow ? I will check this.

satoru-i, 2023-08-22:
Hi @jasvinderkhurana,
I'm using Ubuntu flow(Ubuntu 22.04, Cross-compilation development) found here. 
https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html

jasvinderkhurana, 2023-08-22:
@satoru-i, Thanks for confirming.. I also find this strange because I can see the Qt5Widgets cmake at .../krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/Qt5Widgets/**Qt5WidgetsConfig.cmake** , but the environment is not picking up. I am looking at this and will update you as soon as I get any solution.

jasvinderkhurana, 2023-08-22:
@satoru-i, QT is actict differntly here, though all the dependencies are available in sysroot it is somehow not able to find it. I have found one workaround to fix the cmake find problem, add dependencies(Qt5Core, Qt5Gui, Qt5Widgets) before Qt5 in CMakelists.txt of turtlesim see below:

```
find_package(Qt5Core REQUIRED)
find_package(Qt5Gui REQUIRED)
find_package(Qt5Widgets REQUIRED)
find_package(Qt5 REQUIRED COMPONENTS Widgets)
```

Even after fixing Qt5Widgets error, it gives another issue:

```
aarch64-binfmt-P: Could not open '/lib/ld-linux-aarch64.so.1': No such file or directory
gmake[2]: *** [CMakeFiles/turtlesim_node.dir/build.make:74: include/turtlesim/moc_turtle_frame.cpp] Error 255
gmake[1]: *** [CMakeFiles/Makefile2:598: CMakeFiles/turtlesim_node.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< turtlesim [3.70s, exited with code 2]

```

satoru-i, 2023-08-23:
Hi @jasvinderkhurana, thank you for looking into it.
Applying that workaround to CMakelists.txt of turtlesim worked around the cmake find problem.
But I got a different error than yours. I hope this information is helpful.

```
Starting >>> turtlesim

--- stderr: turtlesim                             
/home/satorui/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/qt5/bin/moc: error while loading shared libraries: libz.so.1: cannot open shared object file: No such file or directory
gmake[2]: *** [CMakeFiles/turtlesim_node.dir/build.make:74: include/turtlesim/moc_turtle_frame.cpp] Error 127
gmake[1]: *** [CMakeFiles/Makefile2:598: CMakeFiles/turtlesim_node.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< turtlesim [7.66s, exited with code 2]
```

satoru-i, 2023-08-24:
Hi @jasvinderkhurana, Thanks for letting me know the workaround. I also built it and was able to run it on the KR260.
With this information, I'm going to try out some user apps using HLS. Thank you. :)

closed 2023-08-24 by satoru-i

miharaka, 2023-08-25:
Hi @jasvinderkhurana,
When I tried cross-compiling packages with PCL(Point Cloud Library) for KR260 in KRS, I got an error related to qt5 similar to turtlesim. What do you have how to solve this problem?

```
colcon build --build-base=build-kr260-ubuntu --install-base=install-kr260-ubuntu --merge-install --mixin kr260 --packages-select pcl_object_detection
```

```
Starting >>> pcl_object_detection
--- stderr: pcl_object_detection                         
CMake Error at /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/Qt5/Qt5Config.cmake:51 (_qt5_Core_check_file_exists):
  Unknown CMake command "_qt5_Core_check_file_exists".
Call Stack (most recent call first):
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/vtk-9.1/VTK-vtk-module-find-packages.cmake:1008 (find_package)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/vtk-9.1/vtk-config.cmake:150 (include)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:270 (find_package)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:319 (find_VTK)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:540 (find_external_library)
  CMakeLists.txt:17 (find_package)

CMake Warning at /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/vtk-9.1/VTK-vtk-module-find-packages.cmake:1008 (find_package):
  Found package configuration file:
    /home/miharaka/krs_ws/install/../acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/Qt5/Qt5Config.cmake
  but it set Qt5_FOUND to FALSE so package "Qt5" is considered to be NOT
  FOUND.  Reason given by package:
  Failed to find Qt5 component "OpenGL" config file at
  "/home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/Qt5OpenGL/Qt5OpenGLConfig.cmake"
  Failed to find Qt5 component "Widgets" config file at
  "/home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/Qt5Widgets/Qt5WidgetsConfig.cmake"
  
Call Stack (most recent call first):
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/vtk-9.1/vtk-config.cmake:150 (include)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:270 (find_package)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:319 (find_VTK)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:540 (find_external_library)
  CMakeLists.txt:17 (find_package)

---
Failed   <<< pcl_object_detection [0.94s, exited with code 1]

Summary: 0 packages finished [1.30s]
  1 package failed: pcl_object_detection
  1 package had stderr output: pcl_object_detection
```