Xilinx/KRS #95: CMake Error : Unknown CMake command "_qt5_Core_check_file_exists".

miharaka, 2023-08-25:
When I tried cross-compiling packages with PCL(Point Cloud Library) for KR260 in KRS, I got an error related to qt5 similar to turtlesim. 
package : https://github.com/klintan/ros2_pcl_object_detection

I used this ( https://github.com/Xilinx/KRS/issues/94 ) as a reference, and add "find_package(Qt5○○ REQUIRED)" to CMakeLists. but I cant solve this problem.
Does anyone have a way to solve this problem?

Required environment : pcl-ros `sudo apt install ros-foxy-pcl*`

```
colcon build --build-base=build-kr260-ubuntu --install-base=install-kr260-ubuntu --merge-install --mixin kr260 --packages-select pcl_object_detection
```

```
Starting >>> pcl_object_detection
--- stderr: pcl_object_detection                         
CMake Error at /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/Qt5/Qt5Config.cmake:51 (_qt5_Core_check_file_exists):
  Unknown CMake command "_qt5_Core_check_file_exists".
Call Stack (most recent call first):
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/vtk-9.1/VTK-vtk-module-find-packages.cmake:1008 (find_package)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/vtk-9.1/vtk-config.cmake:150 (include)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:270 (find_package)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:319 (find_VTK)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:540 (find_external_library)
  CMakeLists.txt:17 (find_package)

CMake Warning at /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/vtk-9.1/VTK-vtk-module-find-packages.cmake:1008 (find_package):
  Found package configuration file:
    /home/miharaka/krs_ws/install/../acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/Qt5/Qt5Config.cmake
  but it set Qt5_FOUND to FALSE so package "Qt5" is considered to be NOT
  FOUND.  Reason given by package:
  Failed to find Qt5 component "OpenGL" config file at
  "/home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/Qt5OpenGL/Qt5OpenGLConfig.cmake"
  Failed to find Qt5 component "Widgets" config file at
  "/home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/Qt5Widgets/Qt5WidgetsConfig.cmake"
  
Call Stack (most recent call first):
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/vtk-9.1/vtk-config.cmake:150 (include)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:270 (find_package)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:319 (find_VTK)
  /home/miharaka/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:540 (find_external_library)
  CMakeLists.txt:17 (find_package)

---
Failed   <<< pcl_object_detection [0.94s, exited with code 1]

Summary: 0 packages finished [1.30s]
  1 package failed: pcl_object_detection
  1 package had stderr output: pcl_object_detection
```

satoru-i, 2023-08-28:
Hi @miharaka,
I added the following find_package before find_package(PCL REQUIRED) in CMakeLists.txt.
```
find_package(Qt5Core REQUIRED)
find_package(Qt5Gui REQUIRED)
find_package(Qt5Widgets REQUIRED)
find_package(VTK REQUIRED)
```

After that, a python error appears, but the QT5 error seems to have been avoided.
```
CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Python3_INCLUDE_DIRS Interpreter
  Development.Module) (found suitable version "3.10", minimum required is
  "3.10")

      Reason given by package: 
          Interpreter: Cannot run the interpreter "/home/satorui/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/bin/python3"
```
I hope this helps.

miharaka, 2023-08-31:
Thanks for the solution. I added its idea, so I got same error.

But its error after adding some "find_package" occurs  at "find_package(VTK REQUIRED)". So I think that it is unclear whether it can solve the PCL error problem.

Therefore, I built the following "find_package" excluding VTK.

```
find_package(Qt5Core REQUIRED)
find_package(Qt5Gui REQUIRED)
find_package(Qt5Widgets REQUIRED)
```

 As result, the following:
```
CMake Error at /home/miharaka/111krS_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:63 (message):

  visualization is required but vtk was not found
```
I read several columns about this error, but I don't know how to solve it.

Also, if it is Improved by adding "find_package(VTK REQUIRED)", I would like to know the solution to the subsequent error "Python3 Development.Module". (I could remove "Python3 interpreter" error.)

Could anyone give me some suggestions to solve these problems?

jasvinderkhurana, 2023-09-14:
@miharaka, were you able to get passed through this error?

miharaka, 2023-09-14:
@jasvinderkhurana , I dont solve this error. The situation remains the same.

jasvinderkhurana, 2023-09-18:
@miharaka, I have tried to check this issue in last few days, but seems like there is some dependencies missing in sysroot when cross compiling QT and VTK packages via KRS. I tried to add the dependencies like pcl_ros by changing sysroot customize patch file at krs_ws/src/firmware/acceleration_firmware_kr260/acceleration_firmware_kr260/patches/customize.patch

```
 18 +    local key_hash=C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
 19 +    local pkgs=(
 20 +        ros-humble-desktop-full ros-humble-image-common
 21 +    )
```

changed to ros-humble-desktop-full to install most of the ros2 dependencies in sysroot. 

I still get issues after doing that, but I will continue to debug more into this.

As a quick workaround I have tried to build the package directly on **KR260 board**, just install ros-humble-desktop-full and build pcl-object-detection package. It builds successfully for me. For now, I request you to please build this package directly on board and let me know if that unblocks you at the moment.

miharaka, 2023-09-19:
@jasvinderkhurana, 

> build the package directly on KR260 board

I tried it, and it was successful.
But I want to build the package in KRS. So I'm grateful that you continue to debug.