Xilinx/KRS #96: AttributeError: 'NoneType' object has no attribute 'split'

ctgulzh, 2023-09-12:
Sorry to bother everyone, I recently encountered a problem while learning KRS: referring to the guidance at [https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html]("https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html",) I built the KRS development environment on a virtual machine , when I proceed to "Select the KR260 firmware artifact and rebuild the accelerator configured for the KR260 build", the following error appears:

[0.356s] ERROR:colcon:colcon acceleration: 'NoneType' object has no attribute 'split'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/command.py", line 522, in verb_main
    rc = context.args.main(context=context)
  File "/home/lzh/krs_ws/install/lib/python3.10/site-packages/colcon_hardware_acceleration/subverb/select.py", line 48, in main
    firmware_options = get_firmware_options()
  File "/home/lzh/krs_ws/install/lib/python3.10/site-packages/colcon_hardware_acceleration/subverb/list.py", line 27, in get_firmware_options
    firmware_options = outs.split("\n")
AttributeError: 'NoneType' object has no attribute 'split'

So,how to solve this problem? I referred to a previous answer: [https://github.com/Xilinx/KRS/issues/30]("https://github.com/Xilinx/KRS/issues/30") but failed to solve the attribute error problem by following the above method.

jasvinderkhurana, 2023-09-14:
@ctgulzh, I have not encountered this problem, can you share the complete steps to reproduce? Just a quick check, have you sourced the environment properly i.e., "source install/setup.bash" ?

ctgulzh, 2024-07-05:
Yes I did "source install/setup.bash"