Xilinx/KRS #97: Cross compilation on Ubuntu 22.04 fails with not finding rmw_cyclonedds_cpp

samlei-research, 2023-10-10:
## Followed steps
I followed these instructions, previously the instructions for KRS installation:
https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html

The native installed ros instance and the local krs_ws are sourced.

It fails when executing:
` colcon build --build-base=build-kr260-ubuntu --install-base=install-kr260-ubuntu --merge-install --mixin kr260 --cmake-args -DNOKERNELS=true` 

## Error
Could not find a package configuration file provided by
  "rmw_cyclonedds_cpp" with any of the following names:

    rmw_cyclonedds_cppConfig.cmake
    rmw_cyclonedds_cpp-config.cmake

  Add the installation prefix of "rmw_cyclonedds_cpp" to CMAKE_PREFIX_PATH or
  set "rmw_cyclonedds_cpp_DIR" to a directory containing one of the above
  files.  If "rmw_cyclonedds_cpp" provides a separate development package or
  SDK, be sure it has been installed.

It does not find rmw_cyclonedds_cpp and its configuration even its installed.

jasvinderkhurana, 2023-10-12:
@samlei-research , 

Thanks for reporting!

Can you please check if you have exported RMW_IMPLEMENTATION on your x86 machine. If yes, please use a different terminal and ensure that the RMW_IMPLEMENTATION is not set while cross compiling. Also, please delete your build and install folders before rebuilding it.

Let me know if that works..

samlei-research, 2023-10-17:
@jasvinderkhurana 

Finally got it working. Issue can be closed. 

It was just a problem with the underlying ROS workspace.

closed 2023-10-17 by samlei-research