Xilinx/KRS #98: Qt5Widgets issues in KV260

mvgianel, 2023-10-17:
I am encountering the same error as the one explained here: https://github.com/Xilinx/KRS/issues/94. I am using Ubuntu 22, but working with the KV260 board -firmware https://github.com/ros-acceleration/acceleration_firmware_kv260/releases/download/v1.1.1/acceleration_firmware_kv260.zip- and I am trying to compile the slam_toolbox package (https://github.com/SteveMacenski/slam_toolbox.git). 
[image]

However, after trying the accepted solution of the previously mentioned issue I still got the same error.

jasvinderkhurana, 2023-11-01:
@mvgianel, I have seen many issues when cross compiling QT applications on x86 machine. Can you please try to compile the same on your Kria board directly. KRS does support your application compilation on board.

mvgianel, 2023-11-01:
@jasvinderkhurana thank you for your reply. I have compiled the application on the KV260, and it works correctly. The issue is that I would like to make an accelerator for the package, and I understood that I couldn't do that on the KV260 since it will require Vitis. That is the reason why I am trying to cross-compile to create the binaries and accelerator to run on the board. Is there another way I should be doing this?

jasvinderkhurana, 2023-11-02:
In that case you are right, if you want to build any accelerator you must use the cross compilation flow on x86 machine. Is there any way that you can decouple your accelerator from QT5? Or you may try hacks that you can copy the built files on KV260 to your x86 machine and then try to resolve all QT5 dependencies.