Xilinx/KRS #99: Dependencies errors and using KV260 as sysroot

mvgianel, 2023-10-18:
I have been running into dependencies problems when trying to build binaries for ROS2 packages, which drove me to try to use my KV260's files as sysroot, but I am also encountering errors when doing that as well.
I tried to build the slam_toolbox package (https://github.com/SteveMacenski/slam_toolbox.git) but got the following error:
[image]
I also tried to build a different package -robot_localization https://github.com/cra-ros-pkg/robot_localization- but I got the following error:
[image]
I then copied the /opt folder in the KV260 to the sysrooot created by the KRS tools and got the following error:
[image]

Since I was getting issues with dependencies that are present in the KV260, it thought it would solve our problems if I used the KV260 as the sysroot. I tried this by mounting the files of the KV260 on my laptop with this command "sshfs -o ro ubuntu@192.168.0.218:/ ./sysroot" and changeing the SYSROOT variable in the kv260.mixin.template file to use the mounted KV260 files. However, when I try to build using that I get the following error:
[image]

Some additional details of the setup:
Host computer has Ubuntu 22.04 operating system and x86_64 architecture. ROS2 Humble was installed along with Vitis 2022.1, as detailed in these instructions: https://xilinx.github.io/KRS/sphinx/build/html/docs/install.html. 
The workspace was also set up as described in the website by following the Ubuntu 22.04 cross-compilation development instructions. However, since the instructions are meant for KR260, the KR260 import was replaced by the KV260 firmware (https://github.com/ros-acceleration/acceleration_firmware_kv260/releases/download/v1.1.1/acceleration_firmware_kv260.zip).
Target is a KV260 Xilinx board with Ubuntu 22.04 operating system. The image for the board was obtained here: https://ubuntu.com/download/amd, and ROS Humble was also installed on this device. The slam-toolbox and robot_localization packages were installed through apt and worked correctly.  

If anyone could help me solve my dependencies problems and/or help me set up the KV260 as sysroot, I would greatly appreciate it.

mohammedrafi-sk, 2023-10-24:
Hi @mvgianel 

You are missing a few packages in the sysroot required to build the robot_localization package. 
The default sysroot of KV260 comes with limited packages. 

After installing the dependencies for robot_localization into the sysroot, I can build the robot_localization package

```
# dependencies for robot_localization
ros-humble-diagnostic-updater \
ros-humble-geographic-msgs \
geographiclib-* \
libgeographic* \
python3-geographiclib
```

[image]

mvgianel, 2023-10-24:
Hi @mohammedrafi-sk 

Thank you for helping me out. I have the packages installed on my laptop, but I don't know how to install them on the sysroot. I have been copying the folders from the KV260 to the sysroot but that doesn't work for all the dependencies. Could you tell me how you installed the dependencies directly into the sysroot?

mohammedrafi-sk, 2023-10-25:
Hi @mvgianel 

The easiest way is to use the shell script ```build-sysroot.sh``` shipped with sysroot. You can find it in ```~/krs_ws/acceleration/firmware/kr260/sysroots/``` folder. 
To install a package to sysroot, you can use this script as follows
```./build-sysroot.sh --update geographiclib-*  libgeographic*  python3-geographiclib ```

To install ros packages to sysroot with this script, You may need to apply this patch to acceleration_firmware and reinstall the acceelration_firmware for KRS. 
https://github.com/mohammedrafi-sk/acceleration_firmware_kr260/commit/96d099dd9c833734a3c8bb0ec47307532c5e9751

The other way is you chroot to sysroot and install packages on to it. The steps are as follows

```
cd ~/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux
mount -t proc none proc
mount -t sysfs none sys
mount -t tmpfs none tmp
mount -t tmpfs none var/lib/apt
mount -t tmpfs none var/cache/apt
mv etc/resolv.conf etc/resolv.conf.bak
cp /etc/resolv.conf etc/resolv.conf
chroot .
```

These steps will give you a chroot environment with sysroot where you can install packages as you usually do in ubuntu. 
Once the installation is done, exit and chroot and unmount the volumes.

```
exit
umount ~/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/var/lib/apt
umount ~/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/var/cache/apt
umount ~/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/tmp
umount ~/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/sys
umount ~/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/proc
umount ~/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/dev/shm
```

jasvinderkhurana, 2023-11-01:
@mohammedrafi-sk thanks for your response here!

@mvgianel, can you please share if your issue is resolved?

mvgianel, 2023-11-01:
Thank you for the help @mohammedrafi-sk. Sorry it took me a while to reply. I am not familiar with chroot so I wanted to try a few things before posting again. 

Unfortunately, when I tried applying the patch, I keep getting this error:
[image]
even though I sourced the install directory.

When I ran the commands without applying the patch, I was able to install the non-ros packages, but I run into issues using the build-sysroot script or using chroot. In both cases I get an error saying that the ros packages cannot be found. I checked apt-cache and get the following:
[image]

Is there a way to add ros-humble-diagnostic-updater to apt?

mohammedrafi-sk, 2023-11-02:
HI @mvgianel 

I believe the ros2 sources are missing in the kv260 sysroot.

To quickly resolve this, Download the attached zip containing the ros2 apt sources and its key and copy them to the sysroot folders as mentioned below
[ros2_sysroot.zip](https://github.com/Xilinx/KRS/files/13237660/ros2_sysroot.zip)
```
sudo cp ros2.list ~/krc_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/etc/apt/sources.list.d/
sudo cp ros-archive-keyring.gpg ~/krs_ws/acceleration/firmware/kr260/sysroots/aarch64-xilinx-linux/usr/share/keyrings/
``` 

The acceleration_firmware adds these sources during the initial build and deletes them once the installation is done. The patch is supposed to keep these sources as is. but it is failing to apply.  I will check and update it. Meanwhile please continue with this workaround.

mvgianel, 2023-11-14:
Hi @mohammedrafi-sk,

I wasn't able to install the packages after coping the files. However, after I also run these commands
```
mount --rbind --make-rslave /dev dev
sudo chroot . apt update
```
along with the previous mount commands you shared with me, I was able to install some of the missing packages as you can see here: 
[image]
[image]
I also had to install
```
ros-humble-geographic-msgs
libgeographic-dev
libgeographic-19
```
After installing the packages that were giving errors, I get the following when trying to build the accelerator:
[image]

Do you know what could be causing this?

mohammedrafi-sk, 2023-11-14:
I believe you missed the step which is included in the warning note in KRS where we symlink the Python3 library
[image]

mvgianel, 2023-11-18:
Hello @mohammedrafi-sk,

I tried that and got the following:
[image]

mvgianel, 2023-11-29:
After running the following commands, I was able to successfully build:
```
sudo mv /usr/lib/aarch64-linux-gnu/libpython3.10.so /usr/lib/aarch64-linux-gnu/libpython3.10.so_host
sudo ln -s ~/krs_ws/install/../acceleration/firmware/kv260/sysroots/aarch64-xilinx-linux/usr/lib/aarch64-linux-gnu/libpython3.10.so.1.0 /usr/lib/aarch64-linux-gnu/libpython3.10.so
```
[image]

Thank you for your help!

closed 2023-11-29 by mvgianel