Xilinx/KRS #104: Issue when building kernels that use hls cordic functions

samlei-research, 2024-01-24:
I want to use the hls::atan() function for a self-defined ap_fixed datatype (ap_fixed<42,24>).

When using vitis hls directly I can run the synthesis, but if I use the ROS2 integration I get the following error when using colcon build XXX.

```
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/XXX.dir/src/XXX.cpp.o: in function `void cordic_apfixed::cordic_circ_apfixed<49, 3, 1>(ap_fixed<49, 3, (ap_q_mode)5, (ap_o_mode)3, 0>&, ap_fixed<49, 3, (ap_q_mode)5, (ap_o_mode)3, 0>&, ap_fixed<49, 3, (ap_q_mode)5, (ap_o_mode)3, 0>&)':
XXX.cpp:(.text._ZN14cordic_apfixed19cordic_circ_apfixedILi49ELi3ELi1EEEvR8ap_fixedIXT_EXT0_EL9ap_q_mode5EL9ap_o_mode3ELi0EES5_S5_[_ZN14cordic_apfixed19cordic_circ_apfixedILi49ELi3ELi1EEEvR8ap_fixedIXT_EXT0_EL9ap_q_mode5EL9ap_o_mode3ELi0EES5_S5_]+0xfc): undefined reference to `cordic_apfixed::circ_table_arctan_128'
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: XXX.cpp:(.text._ZN14cordic_apfixed19cordic_circ_apfixedILi49ELi3ELi1EEEvR8ap_fixedIXT_EXT0_EL9ap_q_mode5EL9ap_o_mode3ELi0EES5_S5_[_ZN14cordic_apfixed19cordic_circ_apfixedILi49ELi3ELi1EEEvR8ap_fixedIXT_EXT0_EL9ap_q_mode5EL9ap_o_mode3ELi0EES5_S5_]+0x100): undefined reference to `cordic_apfixed::circ_table_arctan_128'
collect2: error: ld returned 1 exit status

```

It seems there is an issue with the correct linking, but I did not figure out how to do it correctly. Has somebody a solution to it?

SuperChamp234, 2024-03-01:
Bumping this, did you find a solution?

samlei-research, 2024-03-04:
@SuperChamp234 

I got it compiled with a workaround. I am using Vitis 2022.1, in the installation folder in `/tools/Xilinx/Vitis_HLS/2022.1/src` there is the `hls_math.zip`. I took the definition of `circ_table_arctan_128[128]` from `hls_cordic_apfixed.cpp` in the extracted hls_math, and put it into `/tools/Xilinx/Vitis_HLS/2022.1/include/hls_cordic_apfixed.h` instead of `extern const ap_ufixed<128,2> circ_table_arctan_128[128];`. 

This solution might not be the intended way of using it, but it works for now.

jasvinderkhurana, 2024-04-15:
I hope the solution shared by @samlei-research is working now..

closed 2024-04-15 by jasvinderkhurana