Xilinx/KRS #107: Unable to find rootfs

Claxl, 2024-11-15:
Hello there!
After following the guide found in the repos, at the build moment
```
colcon acceleration linux vanilla --install-dir install-kr260
```
If I select in colcon the kr260 acceleration it fails due to this error
```
Rootfs at krs_ws/acceleration/firmware/select/rootfs.cpio.gz not found.
```
Is there anyway to fix this without using the workaround proposed at #67 , I guess the error might be the same shown in #59.

Some context on my machine:
OS : Kubuntu 22.04
Dual - Boot
Vitis 2022.1
ROS 2 Humble

My environment is based on the doc ones. 
Thanks for any help.

jasvinderkhurana, 2024-11-29:
@Claxl , Sorry for the delay in reply! 

May I know, do you want to build the KRS stack for Petalinux or Ubuntu, for the Kria board?

Claxl, 2024-12-03:
Hello there!
I'm building KRS for Ubuntu

jasvinderkhurana, 2024-12-11:
@Claxl , sorry for the late reply, I was out to attend ROSCon India!

Seems like there is a missing functionality for "colcon acceleration linux", but I am unable to understand your intention here. can you please provide more details on what you want to achieve. 

Do you want to updated some packages in rootfs which is present at **krs_ws/acceleration/firmware/kr260/iot-limerick-kria-classic-desktop-2204-x06-20220614-78-sysroot.tar.xz** ?

Claxl, 2024-12-11:
Hi @jasvinderkhurana, thank you for your reply.
I would like to build KRS Ubuntu stack for the kr260 but once I build it colcon does not find the rootfs which is not present in /iot-limerick-kria-classic-desktop-2204-x06-20220614-78-sysroot.tar.xz.

jasvinderkhurana, 2024-12-11:
@Claxl , I built the KRS stack today for KR260 and I captured the logs for it, please see attached and compare it with yours. Also can you please share the full logs of your build because if there any failure in any of the previous steps it may lead to errors.

[krs_build_logs.txt](https://github.com/user-attachments/files/18094295/krs_build_logs.txt)

Claxl, 2024-12-11:
There are no significant difference in this build logs, I find the problem during the raw disk image creation, in details at this point of the [0_ros_publisher ](https://github.com/Xilinx/KRS/blob/main/sphinx/source/docs/examples/0_ros2_publisher.md)
```
Let's now create a raw disk image for the SD card with PetaLinux's rootfs, a vanilla Linux 5.4.0 kernel and the ROS 2 overlay workspace we've just created for the KV260:

colcon acceleration linux vanilla --install-dir install-kv260
We're now ready to run it on hardware. For that, we need to flash the ~/krs_ws/acceleration/firmware/select/sd_card.img file into the SD card.
```

jasvinderkhurana, 2024-12-11:
I think I got your point now. The documentation is a little confusing today and needs to be updated, because the steps you mentioned are specifically for Petalinux flow and not for Ubuntu flow. So you don't need to do this if you are using Ubuntu on KR260.

Claxl, 2024-12-11:
I got it, thanks.
Is there any reference I can look at to understand the flow?

jasvinderkhurana, 2024-12-11:
@Claxl , I am working on integration of Vitis vision libraries with KRS and then I will improve the documentation and the complete flow for Ubuntu. Is it possible for you to please wait for a week or so?

Claxl, 2024-12-11:
Yes, of course don’t worry. I’m currently trying to use DFX, so I can wait.

Claxl, 2025-01-14:
Hi @jasvinderkhurana, is there any news on the new documentation? I saw the integration of Vitis vision libraries and some of the workflow on the ubuntu side. I guess these are the main differences from the old workflow?

jasvinderkhurana, 2025-01-20:
@Claxl, Yes I completed the documentation on how to integrate Vitis Vision library with KRS. Yes this is a little different workflow and should be straight forward and easy to replicate. Can you confirm whether you are able to run in that workflow?