Xilinx/KRS #108: 'NoneType' object has no attribute 'split'

6visitor, 2024-12-11:
[image]

jasvinderkhurana, 2024-12-11:
@6visitor , seems like some of the previous steps has failed to build. I have checked it on a fresh install or KRS and I don't see any issue. Please check the attached logs, if possible can you share the complete logs as well.
[krs_build_logs.txt](https://github.com/user-attachments/files/18093096/krs_build_logs.txt)

6visitor, 2024-12-12:
[logger_all.log](https://github.com/user-attachments/files/18107842/logger_all.log)
Thanks! I finish this work on KR260 and a computer based NVIDIA 4090, but the error is same.

jasvinderkhurana, 2025-01-20:
@6visitor, Sorry for the delay in reply! 

I checked the logs, but I couldn't see the error logs. BTW are you able to get passed through this error?